wled/WLED #1219: E1.31 support for relay

CaffeinatedFunctionality, 2020-10-02:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
On and off relay support for E1.31

**Describe alternatives you've considered**
Idk but I'm open to alternatives

**Additional context**
no idea, however I wanted to throw some money your way. Where do i donate?

Thank you for your ideas for making WLED better!

Aircoookie, 2020-10-02:
Hi!
Depending on your sending software, this might already be supported!
How would you like it to work, should the relay cut power when no data is received (timeout) or should the E1.31 software actively control the relay?
For the latter, this is already added! You can select DMX mode `Dimmer + Multi RGB` in Sync settings, then the first channel of the first universe will control the WLED brightness and trigger the relay if set to 0.

Coffee gifts go [here](https://paypal.me/aircoookie) :)

CaffeinatedFunctionality, 2020-10-02:
Money sent man, thanks for an awesome program. I was hoping the latter. I guess I just didn't know that. That seems perfect to me.

CaffeinatedFunctionality, 2020-10-02:
Leaving it open just in case I can't get it working but I will close it when I do.

TheOneCalledScott, 2020-10-15:
Hello,

This is more of a question than an issue but it's regarding the exact thing you two are talking about and since the thread was still open, I thought I'd jump in.  Hopefully that's okay.  I was able to trigger a relay board off D6 on my NodeMCU (GPIO12) following Aircookie's instructions above (thank you).  I'm attempting to use the same NodeMCU to control the addressable LED string and the relay for an AC LED floodlight.  The trouble is that when I turn off the first channel to deenergize the relay, the other channels/lights stop responding.  

I'm guessing this is part of the code since in the manual it says the relay is designed to kill power to the string to avoid energy drain, but wanted to see if there's a way to keep the addressable LEDs on when GPIO12 is off, thus killing the relay but leaving others illuminated.  As a backup, I can set up an second NodeMCU and have one for the relay and one for the string...just thought I'd check first to see if there was another recommendation.

keukenrol, 2021-02-11:
When using RGB strips channels 511 and 512 would be suitable, as these are not used.
In that case both pins can be used for relays. For example: send 255 to channel 511 -> pin 5 goes high. Send 0 to channel 511 -> pin 5 goes low.

jemand771, 2021-06-04:
sorry for reviving this somewhat old issue, but:
I was wondering if this could be implemented using a custom usermod. However, as far as i can tell usermods only support incoming mqtt messages, but not e1.31 messages. It would be awesome to have `Usermod::onE131Message` method that usermods could hook into -> this could be implemented by that.

I don't know how the registration of "e1.31 channel listeners" (imma just call them that) works so maybe it's not possible to add such a thing without major refactoring (if wled only listens for 10 leds = 30 channels, you can't have a relay on channel 31)

For now the dimmer + rgb workaround has to do. I have a motor spinning a prisma attached to my relay pin -> i won't be able to use the nice power-saving feature. Though that's fine by me for now