wled/WLED #1220: Connectivity of ws2811 with wled by adding 74HCT125N

shankerrhce, 2020-10-02:
I have issue with ws2811 led strip.  i see a solution in the issues by connecting the level shifter 74HCT125N .

I am a newbie to electronics.  is there any referral diagram or document which I can follow.

huggy-d1, 2020-10-02:
You are in luck.  You can purchase devices with level shifters integrated into the circuit board.  Some of the better ones have protective circuits and fuses to keep from burning up things like the LED strip, microcontroller, and house.

shankerrhce, 2020-10-03:
I am from INDIA, I don't think I will get it  here

Def3nder, 2020-10-05:
Hi @shankerrhce, just connect the level shifter VCC pin to +5V and the the /OE (output enable) + GND pins to ground.

Then connect the LED pin to 1A and from pin 1Y to the LED data.

Look here:
https://learn.adafruit.com/neopixels-on-raspberry-pi/raspberry-pi-wiring#raspberry-pi-wiring-with-level-shifting-chip-3006459-2

This way the WS2811 will get 5V data rather than 3,3V data.

henrygab, 2020-10-06:
@shankerrhce,
You should get the datashseet for your particular chip.  This will help you understand the names for each pin on your chip.  In particular, many have a `DIR` or direction pin.

To configure the `A` pins as input, sending to the corresponding `B` pin:

| PIN | Connect To |
|-----|-----|
| `Vcc` | +5V |
| `GND` | ground |
| `/OE` | ground |
| `DIR` | +5V |

Also, connect any unused inputs to ground also.  This will allow you to transmit from `A1` to `B1`, `A2` to `B2`, ... `A8` to `B8`.

Connect the pin you are using for LED output (from the ESP module) to a corresponding `A` pin.
Connect the corresponding `B` pin to the LED strip data pin (on the input side, of course).

In other words, other than forcing some pins to either ground or +5V, this chip just sits in the middle of the communication between the ESP module and the LEDs.

closed 2021-02-13 by stale[bot]