wled/WLED #1222: Hideously Geeky...

timmills, 2020-10-04:
I've got a rack at home with a bunch of server and network equipment.

I'd like - (because we all like flashing lights) to stick the LED strips to the rack glass in front of each server and display activity of the device, HD capacity, network activity, CPU usage etc..

I'm thinking MQTT? has anyone else tried something like this? essentially turning a range/segment of LEDs into gauges

Aircoookie, 2020-10-05:
Hi, that is a very cool idea!
To get gauges to work well, I would recommend using the `Percentage` effect, whose intensity (when set between 0-100) will illuminate that percentage of the segment.
MQTT is not (yet) ideal for it, because it doesn't support the JSON API yet, which is the recommended way to manipulate segments (you'll want one segment per gauge)
Let us know how it goes :)

huggy-d1, 2020-10-05:
If you have something that can trigger MQTT messages, can it also trigger a CURL command - and use the JSON API?
I use CURL to access the Govee API which requires sending JSON.  Same thing would work for WLED segment control.

closed 2021-02-21 by stale[bot]