wled/WLED #1224: Custom Start Channels

MrPaulAR, 2020-10-04:
**Is your feature request related to a problem? Please describe.**
Feature request

**Describe the solution you'd like**
I would like to be able to define the start channels in addition to universe.  This is for both e1.31 and DDP based channels.

**Describe alternatives you've considered**
Workaround is to use different universes

**Additional context**
The logic behind this request is I would like to limit the amount of wifi traffic as much as possible.  In my specific situation I have 3 WLED based devices controlling only a few pixels each.  By default a whole universe of data is put on the wire.  I would like to have WLED #1 listen for 1:1-9, WLED #2 listen for 1:10-18, and WLED #3 listen for 1:19-27 MULTICAST e1.31.  If using unicast having this distinction is not a valid request.  

If this works as I expect instead of having universe 1 placed on the wire 3 times there would packet sent to 239.255.0.1 (multicast group for universe 1) that each WLED would listen to but only output ws2811 data for the individual channels within the universe 1.

Thank you for your ideas for making WLED better!

Aircoookie, 2020-10-04:
Hi!
This feature is already available, there is a setting `DMX start address` in Sync settings! It works for E1.31 for now but I will add support to DDP as well :)

MrPaulAR, 2020-10-04:
Thanks.  I missed that option entirely obviously.

closed 2020-10-04 by MrPaulAR