wled/WLED #1225: How to use WLED on an Sonoff basic R3 relay and gipo 16 for LED data

monsivar, 2020-10-05:
I was thinking of running wled from an sonoff basic R3 and using the relay for cutting the power and gpio16 for the LED data. How do i configure this, ore have anyone else an finich .bin for it?

pbolduc, 2020-10-05:
> I was thinking of running wled from an sonoff basic R3 and using the relay for cutting the power and gpio16 for the LED data. How do i configure this, ore have anyone else an finich .bin for it?

You can change the LED pin. You need to define LEDPIN at complication. There is a built-in config `custom_LEDPIN_16` that uses GPIO 16. See WLED_0.10.2_ESP8266_ledpin16.bin

monsivar, 2020-10-07:
Thanks for the reply. tried the WLED_0.10.2_ESP8266_ledpin16.bin, it load on the sonoff, but reset wifi settings and i got an error 2

Anyone used the Sonoff basic R3 for this before?

tilitajustin, 2021-01-28:
same issue here. error 10 and after a while wifi settings reset.

tilitajustin, 2021-01-28:
how can we modifi the LED pin to GPIO 9 0r 10? they are available on Sonoff Basic R3. Maybe it will work this way.

aenertia, 2021-02-20:
IIRC GPIO16 does not allow for interupts on the esp8266 - making it unusable for SPI , GPIO3 is the best pin to use as it is (somewhat) 5v tolerant and isn't sharing the SPI bus for the Memory interface (as 14 often is) but it is also the RX pin by default.

9magic, 2021-02-21:
Hi,

I followed your suggestions to use GPIO03 (RX PIN) on my for my sonoff basic r3.
I compiled firmware. I used esp01_1m_full and LEDPIN=3.
All seems to be working fine.

[esp01_1m_full.bin.gz](https://github.com/Aircoookie/WLED/files/6017899/esp01_1m_full.bin.gz)
[image]

closed 2021-06-30 by stale[bot]