wled/WLED #1226: Remove usermod QuinLED_Dig_Uno_Temp_MQTT ?

pbolduc, 2020-10-05:
Question really for @srg74 ...

There are a couple of usermods for using the Dallas Temperature Sensors on the QuinLED Dig Uno / Quad.

@srg74 wrote [QuinLED_Dig_Uno_Temp_MQTT](https://github.com/Aircoookie/WLED/tree/master/usermods/QuinLED_Dig_Uno_Temp_MQTT) which I think was used as the basis for the [Temperature](https://github.com/Aircoookie/WLED/tree/master/usermods/Temperature) usermod. I about a month ago, I did some improvements to the Temperature usermod. I also have some changes in the pipeline for performance (reduce/eliminate flicker) and to reduce the overhead. 

It can be confusing for users to know which one to use. @srg74 are you ok if we remove the old usermod? You are attributed in the new one.  I would also think to rename "Temperature" usermod to something similar to the original usermod name, maybe `QuinLED_Dig_Temperature` it works for both DigUno and DigQuad, so I am open to new names.

srg74, 2020-10-05:
I'm not original owner of the code I just updated to requirements at the time. Like to me if your code is more recent and more reliable it should be only yours as base code.

huggy-d1, 2020-10-05:
Maybe rename it to QuinLED Dig-X Temp MQTT?
Was it updated to Usermod V2 recently?

srg74, 2020-10-05:
@huggy-d1 It was updated after Aircoookie created temp example Usermod V2.

pbolduc, 2020-10-06:
I will put it on my list of changes. I am waiting for delivery of my ESP32 boards to finish testing my changes on the temperature sensor usermod. One main constraint the new usermod will have is that the temperature sensor is the only 1-Wire device on the bus. With this restriction, we can shave off a lot of execution time since we dont need to address which 1-wire device we are talking to.

srg74, 2020-10-06:
I guess it’s perfect since there’s only one sensor on the board. Will wait for changes to test on my board too.

closed 2021-02-13 by stale[bot]