wled/WLED #1227: MQTT hostname does not seem to be honored

sdc53, 2020-10-06:
Hi,
MQTT does not update if I use a hostname instead of an IP address for the broker.  I have other devices running other firmware that work using a hostname, so I do not believe this issue is on my network.

**To Reproduce**
Use a hostname for the MQTT broker instead of it's IP address.  IP address works, hostname does not.  It does not matter if I manually configure WLED with a static IP address or use DHCP.  

**Expected behavior**
MQTT updates instead of doing nothing

**WLED version**
 - Board: Wemos D1 mini
 - Version: 0.10.2
 - Format: binary

Thanks!

huggy-d1, 2020-10-06:
If you use ping from the MQTT broker to that hostname, does it respond?  I bet not.  You can set the hostname using the local hosts. file on the device running MQTT to associate the hostname with the IP address.  Or, you can configure your router to provide a .local address for the device.

Or, is that what you think should be working, like myWLED.local?

sdc53, 2020-10-07:
This seems related to mDNS not working properly.  It doesn't seem to respond to the broadcasted query for it's hostname  via mDNS, or use mDNS for resolution of the MQTT broker either.  I did some snooping via wireshark, there is no response from wled when trying to ping it's configured hostname:
`809	18.742011	192.168.2.107	224.0.0.251	MDNS	78	Standard query 0x0000 A wled-kitchen.local, "QM" question

812	18.742428	fe80::3446:1b7d:d98d:aebc	ff02::fb	MDNS	98	Standard query 0x0000 A wled-kitchen.local, "QM" question

814	18.745548	192.168.2.107	224.0.0.251	MDNS	78	Standard query 0x0000 AAAA wled-kitchen.local, "QM" question

815	18.745980	fe80::3446:1b7d:d98d:aebc	ff02::fb	MDNS	98	Standard query 0x0000 AAAA wled-kitchen.local, "QM" question`

Perhaps something is up with mDNS, or the hostname config?

huggy-d1, 2020-10-07:
You may not have noticed, but there are 2 places where a name can be entered.  One is the hostname, and the other is, well, not.  https://github.com/Aircoookie/WLED/issues/762

Aircoookie, 2020-10-07:
Yes, indeed, there are 3 "hostnames" *of the WLED installation* that are easily confused:
- The mDNS address in WiFi settings. This allows a client supporting Zeroconf to resove xxx.local to the WLED IP
- The server description in UI settings is also used as the IP layer hostname and displayed e.g. in the router device overview
- The MQTT client ID, which is similar in concept to a hostname but MQTT specific

My understanding of the problem is though that WLED doesn't connect to a broker for you if given its hostname.
For me, either a IP or public hostname URL (e.g. `broker.hivemq.com`) works. However, WLED can't resolve mDNS hosts itself (e.g. if you had a private broker reachable at `mqtt.local`). Maybe that is the issue...

sdc53, 2020-10-07:
I have all 3 of those hostnames set the same. However WLED does not seem to be responding to mDNS queries from other devices to find it, according to the wireshark dump.  I also thought it would use mDNS for resolution of the broker, if it does not that would explain my original issue.

Aircoookie, 2020-10-08:
It _should_ respond to mDNS queries from other devices, offering the `_wled._tcp` service. You could try the WLED app or something like Service browser to confirm the WLED device can't be found.

Ah ok, no, WLED can't resolve mDNS itself, you'll have to use the IP if the MQTT broker hostname can't be resolved by a public DNS server.

jsloyer, 2020-11-04:
maybe its worthwhile notating or some form input validation logic that it has to be a fqdn and not a mDNS name.

sdc53, 2020-11-04:
FYI I've also noticed that the version noted above is appending wled to the dhcp request, and the older version does not - when the hostnames are entered identically (with wled). 
[image]

closed 2021-06-04 by stale[bot]