wled/WLED #1228: Request - Wifi AP list

abyssdj, 2020-10-06:
Just a passing thought, but it would be very handy in some cases to be able to have a list of multiple APs rather than just one, possibly with a HTP type priority, so that nodes will attempt to connect to multiple access points before falling back to WLED's own AP mode. The lights I have built are sometimes run on home wifi, sometimes on a hotspot, sometimes standalone, so it would be really nice to see this integrated some time rather than having to switch the wifi settings each time.

hell1337hound, 2020-10-07:
keeeen

Aircoookie, 2020-10-07:
I agree, this would come in handy for all lights which aren't stationary :)
Closing since we already have #852

closed 2020-10-07 by Aircoookie