wled/WLED #1229: question: can Wled be used to control multiple dumb led strips?

kenmcc, 2020-10-06:
Or is there another way? I want to build a singing pumpkin, using dumb led strings - previously all on/off as dictated by switch on the battery box.
Idea is to use one string for the outline, and a few more for eyes, mouth etc.

What I am I missing???

huggy-d1, 2020-10-06:
WLED can control analog LEDs with the right hardware and compilation settings.  Is that what you mean?  Or do you mean a string of LEDs that are either on or off?

kenmcc, 2020-10-06:
yeah - a string of leds that are either on or off. so rather than using a load of pixels of rgb red strips. I want say a string of 10 white LEDs which are "Left eye bottom" and 5 which are "left eye top". So to "open left eye" i turn on all 15 leds, to "close left eye" i leave "left eye top" lit, and turn off "left eye bottom".
Repeat for right eye, mouth open/closed etc.
Each "set" of leds would need to be controlled as if they were one, so would need to be on an individual pin.

hell1337hound, 2020-10-07:
you can get ws drivers to MOSFETs and then you can drive anything from pixel control

closed 2021-02-14 by stale[bot]