wled/WLED #1233: Gs8608 - auto programs kick in when word "off"

hell1337hound, 2020-10-07:
Hey good sir! I've got some GS led strip and the stupid stuff does auto programs when it doesn't see a signal. how much of a pain would it be to request when you press the "off" for wled it actually sends black rather than no data as a tick box option in the settings?

Thanks!!

Legsmaniac, 2020-10-07:
Just an idea - could you not create a setting with no colours and brightness set absolute minimum and save as a selection? Then use that for off? 🙂

Aircoookie, 2020-10-07:
Alright, I believe we need this setting. While setting black as color is a good workaround, having the power usable is way better for usablility e.g. with automation.
Adding it as a setting is the way to go, WS2812 can flicker if the signal is poor (no levelshifter, large voltage diff), that's why WLED doesn't refresh the LEDs when off. Gs8608 manufacturer didn't knew that 😉

harueg, 2022-01-09:
Hi
Was there any development on this issue ?
How often is " blackout " sent ?  From a short test with a logic analyzer, I noticed, it is sent more than once.
I've stumbled over this chip, when looking for an external IC with higher voltage and high PWM frequency.
Although the timing protocol on the paper is different from the WS chips, it seems to work ?
Is there any other chip out there ?

 btw. we are talking about the GS8206 / GS8208 ?

Thanks

Harald

blazoncek, 2023-05-26:
So called "Off refresh" has been implemented in 0.13. #2259

closed 2023-05-26 by blazoncek