wled/WLED #1235: LED Strip suddenly stopped to working

danieely01, 2020-10-07:
Hello, My LED Strip suddenly stopped to working. 
(I ordered all of these parts: https://tynick.com/blog/11-03-2019/getting-started-with-wled-on-esp8266)
Yesterday when I firstly flashed my ESP8266 it was perfect. I connected to the WI-Fi, It worked with the WLED Application too, the LED Strip's lights worked.( WS2812B LED Strip)  it was fine. Untill today. When I connect my ESP8266 to my LED Strip (WS2812B LED Strip) with jump wires, I see nothing. No lights at all. I plugged the wires to the right places (Power - Vin, Ground - GND and Digital out to the D4.)
I hope my LED strip haven't died after 1 day.

Thank you for your help!

huggy-d1, 2020-10-07:
Do you have a picture of the wiring of the strip to the LED strip?
Curious why it was not still wired up today?  Bad cat?
Does the WLED AP show up as a WiFi network?  Does the dev board LED light up when you apply power?

danieely01, 2020-10-07:
[image]
[image]
[image]
[image]

Yeah, the WLED AP show up as a WiFi network, and It says It's connected and online. On the dev board everything is okay.

huggy-d1, 2020-10-07:
Thanks for the pictures.  Tracing wires I have a bit of confusion.

What happens between the NodeMCU Dupont jumpers and the JST connector?  The NodeMCU does not have enough power to directly power an LED strip.

Is WLED loaded with firmware to send out the data pin set to GPIO2 (D4)?  I ask because it might be possible that jumper is on the wrong NodeMCU pin.  It might be GPIO4 (D2)?

Also, having the strip rolled up like that localizes heat and can burn up the strip.  Limit current to around 25-30% and avoid color "white" and effects that turn all LEDs on at the same time.  When it's unrolled, these limits are not necessary to protect the strip.

If there is an external LED strip power supply, be very careful not to burn up a computer's USB port and _always_ keep the NodeMCU disconnected.

If the NodeMCU was powering the LED strip directly yesterday, it probably cannot do it today (or ever again).

danieely01, 2020-10-07:
Thank for your reply.
I'm not really into this yet, so I have some misunderstanding, I'm really sorry about that.
So, for the first question: I use jump wires to connect them, like this: 
[image]
[image]

For the second question: I downloaded the latest firmware, and flashed to the Dev board.The jumper wire connected to the D4 pin: 
[image]
I don't have any power supply, because the guide didn't mention that. Thank you for the tips!!
What power supply should I buy for these?

closed 2021-02-14 by stale[bot]