wled/WLED #1236: Fireworks 1D Effect only works on 1st two segments

robisoel, 2020-10-08:
**Describe the bug**
The Fireworks 1D Effect only works on the first two segments.  Additional segments turn solid on a single color.  

**To Reproduce**
Create three or more segments from your available LEDs.  Switch to the Fireworks 1D Effect.  First two segments display animations as expected.  Additional segments display a single solid color.  

**Expected behavior**
All segments should work with the Fireworks 1D Effect when it is selected.  

**WLED version**
 - Board: Amica NodeMCU
 - Version:  WLED 0.10.2
 - Format:  Unsure what you mean here.  Flashed the NodeMCU with the 0.10.2 software.  No customization was done.  

**Additional context**
I LOVE the software and all the different effects!  You guys have done an amazing job.  Thank you so much!

Legsmaniac, 2020-10-09:
Hmmmm.  I just tried this on a WeMos D1 with a strip of 128 WS2812B LEDs.  I created four segments of 32 LEDs each segment and all four are running the Fireworks 1D effect perfectly,  As expected.
Did you ensure you checked all 4 segments prior to selecting the Fireworks 1D effect?  🤔

robisoel, 2020-10-09:
Thanks for testing!  That is interesting.  I have done it exactly that same way multiple times with only the first two segments displaying properly.  I do have the 2nd and 4th segments reversed so that the 1st and 2nd segments flow together and the 3rd and 4th segments should flow together as well.  Were you running all segments in the same direction?  I have not tried that, but I can see if that makes a difference.  I would prefer to have the 2nd and 4th segments reversed however as it looks better that way on my roofline.

Legsmaniac, 2020-10-09:
OK, just tested reversing the 2nd and 4th segments, still running Fireworks 1D no problems.

Aircoookie, 2020-10-09:
Sounds like you might have a lot of LEDs and that you are running into the 2048 byte limit I set for effect data allocation on ESP8266. Fireworks 1D needs a lot of memory and once the allocation limit is reached subsequent segments will just display the Solid mode. You could try raising the limit (would need a custom compile), but you might run out of memory and crash.
With an ESP32 this limit is 4x as much.

Hope this is indeed the culprit :)

Aircoookie, 2020-10-09:
Ok, just calculated that one segment will need 960 bytes of memory if >= 160 LEDs per segment, so this indeed seems to match up with the fact that 2 segments are working.

Legsmaniac, 2020-10-09:
I do agree, very odd.  I have played around with various settings but I cannot reproduce the problem you're having.  
Only difference is I don't have an Amica NodeMCU and I'm only using 128 LEDs strip but I honestly cannot see they would make any difference really.  
Do you have another/different Arduino to try?  Or else try clearing and reflashing the Amica?
Hope you either manage to suss it or sort it.   🤞

Legsmaniac, 2020-10-10:
Thanks Christian.  I keep forgetting about memory limitations!  Doh!  My bad.  
You will remember I ran a matrix board with 2040 LEDs powered by a single ESP8266, I had no problems with that but then, I didn't split into segments with Fireworks 1D.  So I'm now wondering just how many LEDs @robisoel is running!  😃

So apart from using an ESP32, another solution could be TWO Amica boards, each running two segments with segment 2 reversed and both boards Sync'ed.     Just an idea.  😉

Aircoookie, 2020-10-10:
Two boards would indeed work 😉 
The reason your huge matrix worked is that Fireworks 1D uses a Spark data structure, and there is one Spark per two LEDs, with 80 sparks maximum for the entire segment. Therefore every segment of 160 LEDs or more uses the maximum of 960 bytes (80 Sparks, each is 12 bytes), while smaller segments use proportionally less. Limiting each segment to 40 sparks would also suffice to make 4 large segments work, but the effect might look less epic 🎆

robisoel, 2020-10-13:
Thank you both for your research, testing, and confirmation of the issue.  I appreciate the help.  I will probably got the ESP32 route.  I didn't see a large variety of those on Amazon.  Do you have any recommendations or referrals as to where to buy a good ESP32?  I've seen that there are different models out there as well (32S, 32D, 32U, etc.)  Any recommendations as to which would be best?

robisoel, 2020-10-13:
Thanks so much!

jezmck, 2020-11-09:
For anyone searching, this issue also affects **Fireworks Starburst**.