wled/WLED #1242: Simplify "device name" across app

tablatronix, 2020-10-10:
I do not know if the server description is anything special but it would be nice to use a single `Device ID`
to set the following

* Server name (title)
* mdns hostname
* apname

huggy-d1, 2020-10-13:
I have been wondering why there are 3 options for a name.
Maybe a checkbox [ ] Single host / ap / server name where "Single" gets replaced by "Different" once it is selected.  The implication is the default setting is [Single] rather than [Different] Host / AP / Server Names

tablatronix, 2020-10-14:
Or just a [ ] [use device name] for apname and hostname

mateuszdrab, 2024-12-24:
I actually would like to keep the Server Name separate from the DNS/DHCP hostname.
Having the MDNS and DNS/DHCP names identical make more sense - Server Name which could be seen as a Friendly Name is usually longer and gets truncated badly in a character limited DNS name.
AP name could be inherited from either.

netmindz, 2024-12-26:
You definitely want to retain the ability for the AP name to have the mac address based suffix

mateuszdrab, 2024-12-26:
> You definitely want to retain the ability for the AP name to have the mac address based suffix

Yup, same with hostname, all I need is that name to be equal to the mdns hostname.

blazoncek, 2025-06-06:
Let me revive tis issue (in regards to #4586).

I was looking at the code for mDNS and hostname and did find it a bit odd.
https://github.com/wled/WLED/blob/a87b562bc27d4caa636233c9bbc35b72be123ae8/wled00/wled.cpp#L466
https://github.com/wled/WLED/blob/a87b562bc27d4caa636233c9bbc35b72be123ae8/wled00/wled.cpp#L658-L670
https://github.com/wled/WLED/blob/a87b562bc27d4caa636233c9bbc35b72be123ae8/wled00/cfg.cpp#L59-L61

I would assume most people would want 2 names: mDNS/DHCP/hostname and device name/UI name.

Currently WLED also stores 2 names `cmDNS` and `serverDescription` but their use is a bit odd. `cmDNS` is used for mDNS but not hostname (which is associated with DHCP registration). Instead `serverDescription` is used as hostname/DHCP registration and UI device name. It is also used in Improv and E1.31.

I think `cmDNs` (a field in WiFi settings) should be used for hostname and have `serverDescription`(field in UI settings) used only for UI/Improv/E1.31.

Another oddity is how current hostname is constructed from `serverDescription`.
https://github.com/wled/WLED/blob/a87b562bc27d4caa636233c9bbc35b72be123ae8/wled00/util.cpp#L118-L139

As for AP name, IMO that should be independent from anything else.

Should we change how hostname/mDNS/DHCP name is constructed?

willmmiles, 2025-06-06:
> I was looking at the code for mDNS and hostname and did find it a bit odd.
> Should we change how hostname/mDNS/DHCP name is constructed?

Oh good, it's not just me - I was definitely surprised that the mDNS name and DHCP hostname were different!

I agree that we should have 3 settings:
- hostname (mDNS and DHCP; character set restricted to what's allowable for DNS)
- UI display name
- AP name

In a clean-sheet design I would suggest:
- If unset, hostname defaults to WLED-abcdef, where abcdef are the last three bytes of the MAC address;
- If unset, UI display name defaults to the hostname;
- If unset, AP name defaults to either hostname or WLED-AP (compile time option)

..and for the latter two cases, we store "unset" explicitly (perhaps as an empty string), so it will functionally track changes to the hostname.

I'd also suggest putting the hostname box at the top of the network settings page, before the wifi SSIDs.  During initial setup, it makes it easy to see (or change) how you'll refer to your device before we lose network connectivity to it.  (I'll admit that I have one long-lived WLED board on my test network with no LEDs actually connected that I principally use as a way to find new boards via the nodes list after I join them to the wifi!)

This would let us leverage the hostname as a "device name" in the default/common state.

willmmiles, 2025-06-06:
I guess I was proposing scrapping the use of `serverDescription` in hostnames altogether.   I was thinking something more like:

```c++
void prepareHostname(char* hostname, size_t max_len)
{
    if (strlen(cmDNS)) {
        strncpy(hostname, max_len, cmDNS);
    } else {
        snprintf_P(hostname, max_len, PSTR("wled-%*s"), 6, escapedMac.c_str() + 6);
    }
}
```

willmmiles, 2025-06-07:
Good point; I'd assumed that would be done in `deserialzeJson()`.