wled/WLED #1243: Adafruit Feather Huzzah32 Pin#

jdanbru, 2020-10-10:
I use nearly all adafruit feather huzzah or huzzah32. I can't seem to figure out what pin assignment i should be using when using the ESP32.

I know on the documentation/wiki it says D2. What pin # is that on the feather huzzah32?
[image]

Aircoookie, 2020-10-11:
Hi, yes, `IO_2/A2_2/GPIO_2` is correct. However it doesn't seem like the board brings that pin out to the header for you to solder to.
I would recommend changing the LEDPIN in `NbpWrapper.h`. If you want to use a binary, let me know and I will make one for a pin the Huzzah32 brings out.

jdanbru, 2020-10-12:
if you could, that would be awesome. I'd have to update the libraries you guys are using which may interfere with another project i'm currently working on in parallel to this.

jdanbru, 2020-10-12:
could you use pin 21?

jdanbru, 2020-10-12:
I ended up making it work. I'd recommend Pin 21

Thank

closed 2020-10-12 by jdanbru

Aircoookie, 2020-10-12:
Awesome, great that it works!
I'm very glad you are happy with WLED!
Coffee gift page is [here](https://paypal.me/aircoookie) :)

jdanbru, 2020-10-17:
> Awesome, great that it works!
> I'm very glad you are happy with WLED!
> Coffee gift page is [here](https://paypal.me/aircoookie) :)

Just sent ya $20.

I appreciate it - https://youtu.be/oZsT9W_Nrz4 

I was able to make that in about 20 days for the kids after seeing another house - now i'm addicted.

Aircoookie, 2020-10-17:
Thank you very much :)

Your light show looks awesome - one could think xLights was involved and that's a whole different beast!

jdanbru, 2020-11-03:
> d think xLights was involved and that's a whole different beast!

i'm constructing a 7' tree with 500 leds using Atuline's fork and using a line-in input with his led matrix visualizations this year, that way it'll be synced with the rest of the house. wife hates me i'm sure.