wled/WLED #1244: How do I delete segments using the JSON API?

DivanX10, 2020-10-10:
How can I delete segments using the json api? I can create several segments, but I can't remove them. What command should I use in json to remove all segments by zone?

huggy-d1, 2020-10-11:
From the wiki, JSON API, this is the text that when searched for delete:

`stop | 0 to info.leds.count | LED the segment stops at, not included in range. If stop is set to a lower or equal value than start (setting to 0 is recommended), the segment is invalidated and deleted.`

So, update the _stop_ value to 0 or -1 (via JSON API).

DivanX10, 2020-10-11:
Thanks. Turn out. Everything works. Just in case, I will post a link, I have described everything in detail there and attached files that can be copied and used for their intended purpose.

https://mjdm.ru/forum/viewtopic.php?f=30&t=7249&p=123641#p123641

closed 2020-10-13 by DivanX10