wled/WLED #1246: Overview of icons available in Wicons woff font?

GrumpyMeow, 2020-10-11:
Hi,

I'm trying to find more suitable icons for sliders.
I tried to decode the font and/or find it somewhere online.
Unfortunatly i failed. 

Do you maybe have something like an overview/reference/hint for me?

Aircoookie, 2020-10-11:
Hi!
To reduce file sizes, this is a custom icon pack that only contains the icons that are used in the UI (+maybe just a few icons more I planned to use at some point)
I'm attaching the latest version of that font, I do believe it has more icons though than the woff currently used in the index but I frankly don't know at this point 😅 

The font was made with the free icomoon.io service, which is great. The icons come from a collection/font called `Material-Icons-Complete-Filled`, though I don't have the source for that anymore...

Hope it will help you at least a little. If you want more icons you'll likely have to replace the base 64 woff with a different version that contains all my icons (but likely with different IDs) plus the ones that you want to add.

[mainui-v1.0.zip](https://github.com/Aircoookie/WLED/files/5361100/mainui-v1.0.zip)

[image]

GrumpyMeow, 2020-10-11:
Thank you for the information. I'm working on the "Sound Reactive" fork and would like to do some "polishing". All the current sliders have the same icon.

closed 2020-10-11 by GrumpyMeow