wled/WLED #1247: Segment LED numbering, random restarts using default segment

ShuffleBit, 2020-10-11:
**The problem (and a ... symptom?)**
The configured number of LEDs is 122. The default segment automatically assigns the segment to LEDs 0-122, which is 123 LEDs (one off). It should be either 0-121, or 1-122 to eliminate the off-by-one error. A possibly related issue is that the D1 Mini will restart after a random time (hours), if "Mirror Effect" is enabled in the default segment and saved, or any other change is made to segments. After restart, only the first LED will be lit and no LED changes can be made until the number of LEDs is changed to a lower number (e.g., 12), then changed back to the correct number (122), at which point, only the default Segment is available and it has been reset to default. **Lowering, saving, then correcting, and saving the LED count must be done before the LEDs will respond again.** Rst won't fix it, power cycling won't fix it.

**To Reproduce**
Configure the correct number of LEDs, choose an effect and palette, change the default segment to mirror effect, save that, and wait. I've observed the restart anywhere from 90 minutes to 4 hours after changing the segment.

**Expected behavior**
Segment start/end LED values should be constrained to the number of LEDs. 
I suspect that the confusion displayed in the segment LED numbering has carried over to the FastLED array, and something just beyond the last element of the array is getting stomped (or something just before the array is being stomped).

**WLED version**
WLED 0.10.2, D1 Mini (ESP8266), Self-compiled, WS2812 strip, 122 LEDs

**Additional context**
I might just track it down and fix it, then merge my fix into newer releases, but heap-stomping sometimes spreads and becomes more difficult to debug, and I'd wanna know.

Thank you for your help!
No, thank YOU! This is soooo much better than the code I hacked together years ago.

Aircoookie, 2020-10-11:
Thank you for the detailed report!

It is normal that the default segment has range 0-122 when you have 122 LEDs. The segment length is actually the correct 122 LEDs, the "Stop LED" of the segment is not included in the range. This is so that `stop - start = len`.
It has confused quite a few users so far, so I'm not sure if I should change the behavior, because it is at times not what you'd expect (e.g. to light the first LED your segment would be 0-1 and if you want to have two back-to-back segments, the second would have the same start LED as the stop LED of the first)

I'm quite sure the crash is a different issue and not related to segments. Does it only happen with the mirroring active? What effect(s) were you using for the crash to occur, and are you using any optional interfaces? (MQTT, Alexa, Home assistant, ...)

The single LED lit on startup is really strange. What preset slot have you saved the config to? You can try going to `[WLED-IP]/json/si` next time it happens, maybe we'll be able to figure something out from that :)

ShuffleBit, 2020-10-11:
Well, you're a speedy maintainer!

A little more info: I did a git clone, selected d1_mini, set the LED pin, turned off what I didn't want e.g., INFRARED, and compiled.

Just my opinion. grain of salt, in one ear, early birds, etc: users seem to adjust well enough to physical addressing (the way computers count, or 0 - _N-1_) or they can adapt to logical addressing (the way people count, using their fingers, Windows, or 1 - _N_), but mixing physical and logical in one interface is going to be a problem because it is both physically and logically incorrect. I knew why you did it, but it presents invalid data. Always err on the side of accurate representation. There are two accurate ways to represent a subset of LEDs (a segment), pick _one_.

It could easily be unrelated, but I hate to pester a maintainer, so I waited to submit this until I had tested that a little: I ran fx 72, pa 20 for 24 hours before stopping it. No problem. I started the test again, having changed the default segment to "Mirror Effect". That was the only change. It ran for 3 hours then restarted as described, one pixel.

And, if it's not the segment code, I've seen no shortage of FASTLED quirks, so I'm not even certain it's WLED, but I suspect it is. And I don't mind chasing it down.

I'll reproduce it and grab the json dump, post it here.

Aircoookie, 2020-10-12:
What LEDPIN are you utilizing? I have a unit with GPIO5 and that has random reboots from time to time as well. GPIO2 is the most stable and therefore is the default. The software bitbang pins (everything but GPIO2 and 3 on ESP8266) work well enough generally, but can cause sporadic issues.

Running FX 72, Palette 20 mirrored for 10 hours counting now, seems fine so far.

I wouldn't really call the 0-N approach _invalid_ (as long as you know the stop LED is not the last LED), but I have to agree that it is _counterintuitive_. A frontend change would fix this for most users, but then we have confusion for API users as for why stop is last +1. So a breaking API change would be in order. I might consider that for 0.11 because starting with that version, presets will be stored as API commands and it will be harder to make changes down the line. We could add an `end` property in its place, deprecating `stop`.

ShuffleBit, 2020-10-12:
That makes a ton of sense. I used the pin where my test strip landed when I jammed its Dupont jumper into the breadboard. Seriously, that was the determining factor. I'm using IO-0 (D3 on the Mini), right next to IO-2 (D4 on the Mini).

Ya know how you always think of what you shoulda done _after_ it's too late? I should have grabbed a json/si before the freeze as a reference. But here's what it is after the freeze (only LED[0] is lit, solid blue, web UI responds, LEDs do nothing):

`{"state":{"on":false,"bri":128,"transition":7,"ps":-1,"pss":33023,"pl":-1,"ccnf":{"min":1,"max":5,"time":12},"nl":{"on":false,"dur":255,"fade":true,"mode":1,"tbri":0},"udpn":{"send":false,"recv":true},"lor":0,"mainseg":0,"seg":[{"id":0,"start":0,"stop":122,"len":122,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,160,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false}]},"info":{"ver":"0.10.2","vid":2010020,"leds":{"count":122,"rgbw":false,"wv":false,"pin":[0],"pwr":0,"maxpwr":0,"maxseg":10,"seglock":false},"str":false,"name":"WLED","udpport":21324,"live":false,"lm":"","lip":"","ws":0,"fxcount":114,"palcount":52,"wifi":{"bssid":"88:DC:96:02:55:14","rssi":-48,"signal":100,"channel":6},"arch":"esp8266","core":"2_7_4","lwip":1,"freeheap":22264,"uptime":4730,"opt":71,"brand":"WLED","product":"FOSS","mac":"5ccf7f5c444a"}}`

But I'm switching to IO-2 now. So maybe this is all moot.

I didn't say the 0-_N_ approach is invalid, I said it was universally incorrect. It's like bestiality on an elevator: wrong on many levels. A kind of technological affront. And, yes, it's a UI (and API) change.

As far as API compatibility goes, as long as a change is made to correct or improve something, even if it breaks backward compatibility, _and_ it's in the change log, I will never complain about it. Not out loud. Not where anyone could hear. I've been using Tasmota as a base for a long time, and there have been a few compatibility breakages in that project, but all of them were improvements or corrections. Hardly anyone complains, and that's a massive install base. **I really like its MQTT implementation.**

I'll get this shifted one pin to the left and start another test (after I grab the pre-test json/si as a reference).

Thank you yet again for this truly wonderful project.

ShuffleBit, 2020-10-12:
As far as the freeze goes, that appears to have been caused by using D3 (IO-0) as the WS2812 data pin. Switching to IO-2, it's been running for 6 hours with no problem. That's a very odd thing all by itself.

closed 2021-02-21 by stale[bot]