wled/WLED #1248: Run weathersketch and wled sketch independently over an esp8266

LAR92, 2020-10-11:
Hello, I'm completely new to the world of programming, I have a nodemcu esp8266 2m ws2812b led strip, a 1.3 inch OLED display and a bme280 sensor, I want to build an infinity mirror and have the led strip for it, I already have one open source sketch showing the weather data time date air pressure humidity air temperature, the esp8266 is connected to the sensor and the screen, this sketch runs perfectly, i want the wled to run independently of the weather sketch, my question is how i can do it, i had too already flashed on the esp8266 everything worked as you can see on the picture, but how can the esp8266 control both at the same time?

[image]
[image]

[image]
[image]
[image]

                          

WEATHERSKETCH:

/**The MIT License (MIT)

Copyright (c) 2018 by Daniel Eichhorn - ThingPulse

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in all
copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
SOFTWARE.

See more at https://thingpulse.com
*/
#include <ESPWiFi.h>
#include <ESPHTTPClient.h>
#include <JsonListener.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>
// time
#include <time.h>                       // time() ctime()
#include <sys/time.h>                   // struct timeval
#include <coredecls.h>                  // settimeofday_cb()
//#include "SSD1306Wire.h"
#include "SH1106Wire.h"
#include "OLEDDisplayUi.h"
#include <Wire.h>
#include "OpenWeatherMapCurrent.h"
#include "OpenWeatherMapForecast.h"
#include "WeatherStationFonts.h"
#include "WeatherStationImages.h"

// Create the Lightsensor instance
#define BME_SCK 13
#define BME_MISO 12
#define BME_MOSI 11
#define BME_CS 10
#define SEALEVELPRESSURE_HPA (1013.25)
Adafruit_BME280 bme; // I2C
//DHTesp dht;
/***************************
 * Begin Settings
 **************************/

// WIFI
const char* WIFI_SSID = "FRITZ!Box 7490";
const char* WIFI_PWD = "Password...";
[... 285 lines cut ...]
  display->drawString(x + 20, y + 34, temp);
  display->setTextAlignment(TEXT_ALIGN_LEFT);
}

void drawHeaderOverlay(OLEDDisplay *display, OLEDDisplayUiState* state) {
  now = time(nullptr);
  struct tm* timeInfo;
  timeInfo = localtime(&now);
  char buff[14];
  sprintf_P(buff, PSTR("%02d:%02d"), timeInfo->tm_hour, timeInfo->tm_min);

  display->setColor(WHITE);
  display->setFont(ArialMT_Plain_10);
  display->setTextAlignment(TEXT_ALIGN_LEFT);
  display->drawString(0, 54, String(buff));
  display->setTextAlignment(TEXT_ALIGN_RIGHT);
  String temp = String(currentWeather.temp, 1) + (IS_METRIC ? "°C" : "°C");
  display->drawString(128, 54, temp);
  display->drawHorizontalLine(0, 52, 128);
}

void setReadyForWeatherUpdate() {
  Serial.println("Setting readyForUpdate to true");
  readyForWeatherUpdate = true;
}

Aircoookie, 2020-10-12:
Hi!
Just judging from the hardware, it should be possible to run this sketch alongside WLED.
This would be possible by converting it to a WLED usermod.
Here is an [example usermod](https://github.com/Aircoookie/WLED/blob/master/usermods/EXAMPLE_v2/usermod_v2_example.h) for WLED to give you an idea.
Making an usermod is quite doable, but not trivial.
Keep in mind to
- put the code to `setup()` that you want to run at boot (display initialization)
- in `connected()` put the code you want to run after WLED has connected the wifi (get weather data)
- in `loop()` put what you put in arduino loop. Make sure you replace `delay()` calls with a timer, you can take a look at the example to see how to achieve this.

Hope it helps! If this is too much work, I would consider using a second esp8266 for WLED, even if you want the light in the same location.

LAR92, 2020-10-12:
Ok thanks, I'll try it, I'll report how it went.

LAR92, 2020-10-12:
When I check the sketch I get the message:

 Arduino: 1.8.13 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\Lauritz\Downloads\WLED-0.10.2\WLED-0.10.2\wled00\wled00.ino:13:0:

wled.h:49:27: fatal error: ESPAsyncTCP.h: No such file or directory

   #include <ESPAsyncTCP.h>

                           ^

compilation terminated.

exit status 1

ESPAsyncTCP.h: No such file or directory

I'm not sure if I did everything right.

LAR92, 2020-10-16:
I have ordered a second esp8266 and will operate the weather station and wled independently.

closed 2020-10-16 by LAR92

Aircoookie, 2020-10-17:
Hi, sorry for not getting back to you earlier!
That's the best solution IMO, now you won't have to mess with one system if the other breaks or if you want to update it.

In case you wondered, that compilation error means that the `ESPAsyncTCP.h` library is missing. You'd need to install all the libraries listed in the main page of the WLED wiki to compile it successfully. I've come to recommend using VS code with platform.io since it saves you that hassle :)

LAR92, 2020-10-17:
Ok thanks 👍