wled/WLED #1250: HA Integration cant change IP

hell1337hound, 2020-10-12:
Hey i've got wled running on my wifi and i static'd the IP different from what it was but i used the integration for HA to detect wled initially but now its moved IP and i cant see a way to update the IP for the HA integration 

Any ideas?

Aircoookie, 2020-10-12:
I'm not an HA expert, but I believe you can somewhere add a new device by IP.
Alternatively, if you reboot WLED, HA might recognize the light on the new IP.
If nothing else helps, you could remove the light from the HA config and it should be automatically re-discovered.

hell1337hound, 2020-10-12:
hey I'm using the integration to do presets so I think if I remove it and readd it I'll loose my scenes 😔

Sent from Nine<http://www.9folders.com/>

huggy-d1, 2020-10-12:
So, do you want to get the ip address back to what your integration is configured to, or do you want the integrations updated to the ip address your device running wled has right now?  Or, something else?

hell1337hound, 2020-10-13:
I'd like to get the integration updated to the ip address your device running wled has right now

Sent from Nine<http://www.9folders.com/>

huggy-d1, 2020-10-13:
If I were to guess, you will probably have to find some configuration .yaml file for your WLED node(s) and manually edit the IP address.

Either that, or delete the node and have HA auto-detect WLED installs again.

sergge1, 2020-10-18:
@huggy-d1 @hell1337hound @Aircoookie 

I understood as I had the same problem as @hell1337hound  describes.
I resolved it by assigning a static IP adress to my two WLEDs. However HA integration should work in a little bit different way.

1. Added HA integration for WLED - Ok
2. HA discovered WLED - Ok
3. HA manages WLED through integration - Ok
----------- thats all for now -------
Now how integration works further
4. Reboot WLED and 
4.1. WLED got new IP address from the router
4.2. HA does not found the integration from p.2 so can't manage WLED.
4.3. HA discovers new WLED instance with new IP address  (p.4.1.) and could be added as new WLED integration.

How it should work:
2.1. HA should write MAC address of the WLED device to the HA's WLED integration;
4. Reboot WLED and
4.1. WLED got new IP address from the router
4.2. HA's WLED integration found the new WLED device and read its MAC address;
4.3. HA's WLED integration checks the MAC address from p.4.2 with those from p.2.1.
4.3.1. If MAC 4.2 = MAC2.1. - works with WLED as predefined WLED integration (equal mac means that newly discovered WLED device had been integrated before).
4.3.2. If MAC 4.2. =/= MAC 2.1. - HA makes new WLED integration (non equal mac means that new WLED device had been added to the network).

hell1337hound, 2020-10-18:
yep I completely agree. Mac address based assigning

Sent from Nine<http://www.9folders.com/>

sansillusion, 2020-10-18:
Yes that would be great. I too had that issue at one point. Had to redo a
load of automations and scripts.

hell1337hound, 2020-10-18:
yeah scenes were my issue

Sent from Nine<http://www.9folders.com/>

closed 2021-03-19 by stale[bot]