wled/WLED #1251: When power is applied, there are two configurable behaviors: On / Off. I'd like a 3rd option: Saved.

ShuffleBit, 2020-10-12:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
There are two configurable conditions for determining a module's behavior when power is applied: On or Off. I'd like a third behavior when power is applied to the ESP: whatever the modules state was when it lost power. Call it "Saved", or "Last Used". This can be implemented by using one bit in EEPROM: set the bit when a Power On command is applied, or reset the bit when a Power Off command is applied. When the configured state is "Saved", that bit is read and the state set to that value.

The benefits of this are clear for anyone living where utility service is dodgy, e.g., CA. I don't live in CA, but we routinely have power outages too. Usually a few seconds, but that's enough to turn WLED off.

**Describe alternatives you've considered**
Adding it myself. I think others would appreciate this feature. Otherwise, it wouldn't be a common option in other systems and firmware.

**Additional context**
Imagine a home theater, subtly lit by LED strips that turn on when pause or stop is pressed on a remote. Ahhh. Now, imagine there was a brief power failure, all the electronics are now off, including the LED strips. Nothing can be seen, but a muffled curse is heard as a toe meets a table leg. Another, as a face meets a wall. One bit in EEPROM could have spared that toe and face.

Thank you for your ideas for making WLED better!
No, thank YOU!

Legsmaniac, 2020-10-13:
+1
I like this idea.
Other devices have a memory (eg Sonoff) so why not WLED too?  🙂

Aircoookie, 2020-10-13:
Hi!
The problem with adding this feature so far was that EEPROM is not suitable in the long term for this application.
The "EEPROM" on ESP8266 is actually just an abstraction layer of a single flash memory block. Writing to that repeatedly will wear it out typically after 10.000 write cycles, with very cheap flash (like it tends to be used in very cheap microcontrollers) it may be as low as 1000 cycles before potentially bricking the ESP. Using it for presets and settings was therefore fine, but writing to it on every state change is not.

With the new filesystem support in v.0.11 this changes however. We have wear leveling, which means the writes will be (semi) distributed to the different flash blocks so that each one gets less worn out. With that this becomes more doable, but there is another issue. While the LittleFS filesystem is not **SLOW** it is _slow_ and will block for a half second or so, causing the LEDs to not update for that period. Again, tolerable for saving presets but not really for state changes. Still, I will give adding that feature a try and if the additional latency of saving the state is not too bad I will add it as an option 🙂

ShuffleBit, 2020-10-13:
Yes, I've been lucky. Writing toggle states to EEPROM twice a day for years... Drunks and idiots get a pass, they say. It's like 'they' know me...

A 200-500ms hesitation at toggle doesn't cause headaches for me. I'll be surprised if it bothers anyone. While running, sure, but this isn't doing that during operation. I'm skeptical of the flash chips some of these things use. Even if it said "Winbond", I'd assume it's fake.

I own some genuine ESP32-WROVER modules, but they're somewhere in or over Texas last time I checked. I'm going to fit one with a microSD card because I gotz me a idear.

I'll do some timing tests to see how long it takes to serialize various sized stuff to microSD, trying a few different microSD cards. Most decent microSD cards wear-level on their own, and an unrecoverable corrupted sector on microSD is mappable. I have to make PCBs, so it'll be a week or so before I can design/run a test. I'll post a comment with results, so leave this open a while longer, please.

Legsmaniac, 2020-10-13:
> Writing to that repeatedly will wear it out typically after 10.000 write cycles

If the On/Off State was written to EEPROM (say) twice a day, every single day, 10,000 write cycles would last 13.7 years!  Even a 1000 write cycles on a cheap-and-cheerful unit would last 1.4 years in the above scenario.  And given how cheap most Arduino unit are nowadays, I'd say it's worth it to have the On/Off State.  If the Chinese can do it with their cheap and cheerful SonOff devices...........    😊

I guess it also boils down to just how many times this particular State is likely to be written to.  Each time WLED is turned On/Off - sure, assuming the WLED is physically turned Off.  In my case, my most-used case, my Xmas lights are only going to be in use for just over a month once per year, which means in the above scenario, write states should last a lifetime.  A human lifetime.  Not only that, some of my Xmas lights are turned off automatically by cutting the power, just like a power cut.  (So the State option could be very useful to me.)  This means there would be NO write state written to EEPROM therefore most such units would theoretically last forever.

Long story short..........
Can we please pretty please have On/Off States ?  😁

Aircoookie, 2020-10-17:
That makes sense, if the on/off state is only written likely twice a day, it will take an acceptably long time until the flash is worn out. But since there will be applications where the state is changed more frequently, it will be optional.
I not making changes to EEPROM anymore as I am currently working on transitioning to filesystem - but once that is done, I am going to add this feature - either off (no states saved), On/Off (state recalled), or Full (entire configuration recalled)

pfeerick, 2020-10-23:
It might be worth looking at how Tasmota implements this, as they have the 'saved' model for the power state of relays, etc... I'll link to the code if I remember to come back to this in a timely fashion...

ShuffleBit, 2020-10-23:
Full description of Tasmota PowerOnState command

⁣Thanks,
Shuff

pfeerick, 2020-10-25:
The actual implementation of the flash wear leveling seems to be in

(SettingsSave) https://github.com/arendst/Tasmota/blob/e176aaf2886d5fc6de1f6ec1fb7c44e45c06306e/tasmota/settings.ino#L515-L574 

and 

(SettingsLoad) https://github.com/arendst/Tasmota/blob/e176aaf2886d5fc6de1f6ec1fb7c44e45c06306e/tasmota/settings.ino#L576-L614

When you look at the [flash memory map](https://github.com/arendst/Tasmota/blob/e176aaf2886d5fc6de1f6ec1fb7c44e45c06306e/tasmota/settings.ino#L134-L153), 0x000F4000 to 0x000FA000 is a rotating settings area. Since Jason is looking at WLED atm, it would be worth seeing if he can help with shoe-horning it into WLED...

LineF, 2021-12-24:
Hello,
any news about this feature?
I would also like to restore to the previous state as my led strip with controller will be switched on/off with a mains switch.

Suggestion:
The "last state" wouldn't have to be saved immediately after the state change but e.g. after one minute.
So if someone plays around with effects, not each state would be saved to eeprom but only the last one (after a minute)...

blazoncek, 2021-12-24:
It is implemented in Auto Save usermod.

closed 2021-12-24 by blazoncek