wled/WLED #1252: Alexa Voice Assistant are not connecting

Stoppersocke00, 2020-10-13:
I have flashed my ESP with the Newest Binary of WLED (the Audio Reacitve one) and tried to connect my Alexa with it. The Alexa app can't find any device so I looked if I have all seetings correctly. I have Alexa emulated device on, have setten a Name for the device and have activated UDP in my Router. But still it won't work. Do anyone know what I'm doing wrong? 

 - Board:  ESP32 Wroom Devkit V1
 - Alexa Voice Assistant: Echo Dot V1 Gen.1
 - Version: WLED 10.2 (build 2009030) 
 - Format: binary
 - Wireless AP:  Raspberry Pi with hostapd and dnsmasq

I heared of the Problem with the Echo Dot V1 Gen.1 wich don't use UDP for searching devices. But I have the latest Version of WLED the bug should be fixed. Or not?

Thank you for your help!

huggy-d1, 2020-10-13:
There are 3 places to assign a "name" to WLED.  One of them is for Alexa to use. Try setting all 3 to the same thing.

Stoppersocke00, 2020-10-13:
All of the names are still the same so it can't be the issue. But nevertheless thank you for your tip :D

huggy-d1, 2020-10-13:
Does this issue https://github.com/Aircoookie/WLED/issues/931 explanation by @Aircoookie help?

Stoppersocke00, 2020-10-14:
So bc I use a Physical alexa device (Echo Dot) it can't be the problem.

huggy-d1, 2020-10-14:
Do you have any screenshots of the configuration, or a json backup you can post?  Maybe there is something we would ask or suggest if we could see what you see.
Of course, review this github issues tab for "Alexa" by adjusting the filter: is:issue is:open is the default for me. If you used is:issue Alexa as a filter, all issues with Alexa mentioned should appear.
You cannot be the only one with the exact same (or very similar) problem.

derekpurdy, 2020-10-15:
I have the new Echo Dot, it won't auto discover my 2 wled devices and it also won't see them when I scan for hue devices.

D1 Mini

brunorichter, 2020-10-26:
I have a Echo Dot 3rd Gen and can't detect the ESP32. I tested it with code that worked a while ago but now it doesn't work. The same ESP, the same wifi, modem, ....
The code and the ESP is working, but Echo dot don't detect. I checked via serial

PeteBurgess, 2020-10-27:
I compiled with debug as I've having a similar issue. When Alexa runs discovery, the debug console shows :
`Not-Found HTTP call:
URI: /api/2WLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBQr/lights
Not-Found HTTP call:
URI: /api/2WLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBQr/lights/100153665
Not-Found HTTP call:
URI: /api/2WLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBQr/lights`

brunorichter, 2020-10-27:
> I compiled with debug as I've having a similar issue. When Alexa runs discovery, the debug console shows :
>
> Not-Found HTTP call:
> URI: /api/2WLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBQr/lights
> Not-Found [...]

With debug here is the same!
```
Responding search req...
# Responding to description.xml ... #

Send setup.xmlNot-Found HTTP call:
URI: /api/2WLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBQr/lights
Body: 
AlexaApiCall
ok
l0
lAll
Not-Found HTTP call:
URI: /api/2WLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBQr/lights/74994945
Body: 
AlexaApiCall
ok
l74994945
1
Not-Found HTTP call:
URI: /api/2WLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBQr/lights/74994946
Body: 
AlexaApiCall
ok
l74994946
2
Not-Found HTTP call:
URI: /api/2WLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBQr/lights/74994947
Body: 
AlexaApiCall
ok
l74994947
3
Not-Found HTTP call:
URI: /api/2WLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBQr/lights
Body: 
AlexaApiCall
ok
l0
lAll
```
how do i solve this?

PeteBurgess, 2020-10-28:
Dug a little further. EspAlexa will sometimes respond to the call for the description.xml but calling for setup.xml after that fails with “not found”. Don’t know if that’s by design or not.

peteakalad, 2020-11-04:
I had same issue for some time, but managed to resolve it by leaving my wled on, turning off my echo show and turning it on again.  Once it booted I immediately did a 'alexa discover devices' and it found it found the light.

Also I'd seen mentioned on older issues that upnp needed to be on the on router - in my case this worked while it was disabled (for security reasons dislike upnp).

Psyobilin, 2020-11-10:
> I had same issue for some time, but managed to resolve it by leaving my wled on, pulling the power on my echo show and plugging it in again. Once it booted I immediately did a 'alexa discover devices' and it found it found the light.

Worked here as well, after trying evrything possible! Thank you @peteakalad

brunorichter, 2020-11-10:
Guys, here don't work. I leave my led on in pwm mode on tasmota. I pulled the power on my echo dot and plug again. The light started to turn. After blinked i said "alexa discover devices" but not found. Am I doing it right like you? @peteakalad @Psyobilin

peteakalad, 2020-11-10:
Yes. You have ‘emulate Alexa device’ checked ok? If so also try changing the Alexa invocation name to something else, reboot WLED, then hard power cycle your dot as above and do discover devices.

Beliar-666, 2020-11-17:
Same Issue here. Alexa won´t find my WLED Devices. The older ones still present and work but if i try to add new Devices Alexa won´t find them.

derekpurdy, 2020-11-17:
What device type are you supposed to use? Hue v1 Hub?

Beliar-666, 2020-11-17:
I dont have any hue Hub. Cause i dont need one.

derekpurdy, 2020-11-17:
> I dont have any hue Hub. Cause i dont need one.

how do you kick off the detection in the alexa app then? maybe i'm missing something here.. do you add device and choose a hue device? or should it just auto detect it?

Tesafilm04, 2020-11-19:
Hi, I have the same problem lately and can't find my new Wled devices with Alexa. Everything else is going great. Any idea what the problem is and how I can get around it? Really cool project what you have built!

huggy-d1, 2020-11-19:
I believe an alexa enabled device needs to _discover_ WLED is running.
Others with Echo DOTs have reported success by simply power cycling an Echo DOT and asking it to discover new devices after it powers up.

Tesafilm04, 2020-11-19:
I have a second generation Echo dot and I have also tried it. So plug it out and then back in again, if this is power cycling the echo dot

Tesafilm04, 2020-11-19:
It worked

huggy-d1, 2020-11-19:
@Tesafilm04 what worked?

Tesafilm04, 2020-11-19:
Alexa found the device. I dont know why it worked now and not three days before, but Alexa found my two Wled devices.

brunorichter, 2020-11-19:
> Alexa found the device, after the power cycling. I dont know why it worked now and not three days before, but Alexa found my two Wled devices.

I will try again

spangles80, 2020-11-23:
Cheers for all the advise and ideas. Unfortunately my WLED's can't be found, but will try every day as it seems very hot and miss

baycom, 2020-11-23:
Same problem here. Echo restart does not help at all. Alexa support seems to be really broken.

Tesafilm04, 2020-11-23:
if it then works, then the whole thing goes super ;-)

zachmatts10, 2021-04-01:
Hi, is there any tried and true method for reconnecting these devices. All of my WLED devices stopped working with my Alexa, so I removed them and am now unable to discover them again. Is there a way to observe via a serial monitor for debugging? (I am using the provided .bin files to flash to the ESP8266 if that's helpful info)

kuno23, 2021-05-05:
I have same problem, my device stopped working with alexa, i removed it and now alexa is not able to find my device. I tried the power cycle solution, but nothing helps, so now i cannot use my wled devices with alexa. Maybe that was an amazon update for devices, and problem is there. Because i cannot find any working solution for now.

etlon, 2021-07-02:
same, tried everything, my alexa isn't finding anything

a-jey, 2022-04-15:
Same here,
Are there any news on bugfixing?

Thank you.

etlon, 2022-04-16:
@a-jey for me it was me using the firetv cube, instead of an actual alexa. Bought the echo and it worked like a charm.

softhack007, 2023-10-02:
Stalebot is right, this issue seems to be stale for very long now. Closing.

closed 2023-10-02 by softhack007