wled/WLED #1256: Add new Effect

mehranviva, 2020-10-14:
hi,

can you help me write a code for adding new effect
I have already written this code for analog LEDs, which I will put here
`for(uint8_t  j=0;j<LED_COUNT;j++)
            {
                  for(uint8_t i=0;i<LED_COUNT-j;i++)
                  {
                     digitalWrite(led[i],HIGH);
                     delay(D1);
                     digitalWrite(led[i],LOW);            
                  }
               digitalWrite(led[LED_COUNT-j-1],HIGH);  
            }
          for(uint8_t j=LED_COUNT;j>0;j--)
            {
                  for(uint8_t i=LED_COUNT-j;i>0;i--)
                  {
                     digitalWrite(led[i],HIGH);
                     delay(D1);
                     digitalWrite(led[i],LOW);            
                  }
               digitalWrite(led[LED_COUNT-j],LOW);
            }`

my main problem is delay
you can see it in this link
[https://www.youtube.com/watch?v=kA-WaNq4nsQ](url)
0:29-0:42

huggy-d1, 2020-10-14:
Replace your delays with a time difference comparison.  Save the time something happened, then each time WLED loops, compare the difference between current time and saved time.  If it is >= required delay time, make something happen and set the saved time to current time.

Repeat for all the delays.

That's the "I have blinders on" approach.

Better approach is to compare existing effects to the visual effect you want and have coded for solo operation.
Open the code for that effect and see how it works in WLED.
Make a copy of that effect, and try one change to see if you can get the start of the effect to match what you are after.
Repeat until the entire effect is done.

You may be a super-coder and not someone that says "I just want to keep smashing at the code until it bends to my will". If you are a super-coder, please ignore everything I just wrote!

huggy-d1, 2020-10-15:
https://github.com/Aircoookie/WLED/blob/master/wled00/FX.cpp
Pick one of the effects close to the effect you want, and see how it works in wled.
Then make a new effect (copy that effect or modify the code for that effect).  Then using VS Code / PlatformIO, build, test, edit and repeat until it does what you want.

fishbone-git, 2020-10-15:
Also have a look here for a very simple list of where to look for the lines to change to add an effect

https://github.com/Aircoookie/WLED/wiki/Add-own-functionality#create-custom-effects

mehranviva, 2020-10-22:
Any idea for writing this effect?

`for(uint16_t i=0; i<SEGLEN; i++) {
    for(uint16_t j=0; j<SEGLEN()-i; j++) {
      setPixelColor(j-1, SEGCOLOR(1));
      setPixelColor(j,SEGCOLOR(0));
      delay(wait);
    }
  }
`
My big problem is delay

Aircoookie, 2020-10-23:
Yep, delay won't work as it will block everything (including actually updating the LEDs)
The way the effect function works is that WLED will wait the time you have the function return before calling it again. That in turn causes the for loops to not work as we need to spread them between multiple calls to the effect function.

This would be the WLED-compatible equivalent to your code, although I haven't tested it:
```cpp
uint16_t WS2812FX::myEffect() {
  //instead of i and j we use SEGENV.aux0 and SEGENV.aux1 as these are preserved between function calls
  
  
  if (SEGENV.aux1 > 0) setPixelColor(SEGENV.aux1-1, SEGCOLOR(1)); 
  setPixelColor(SEGENV.aux1  , SEGCOLOR(0));
  
  SEGENV.aux1++;
  if (SEGENV.aux1 >= SEGLEN - SEGENV.aux0) {
    SEGENV.aux1 = 0; SEGENV.aux0++;
    if (SEGENV.aux0 >= SEGLEN) SEGENV.aux0 = 0;
  }
  
  return wait;
}
```

mehranviva, 2020-10-25:
Thank you, it worked great and without any problems
Can you help me to set the speed and intensivity with bargraph, or give me an example

Aircoookie, 2020-10-25:
You can access the slider values for speed and intensity (range 0-255) with `SEGMENT.speed` and `SEGMENT.intensity`. I don't really know what intensity would be used for in this particular effect (intensity is kind of general purpose, what it does depends on the effect), but an easy example for getting speed to work would be replacing the last line with e.g. `return (255 - SEGMENT.speed) + 5`

mehranviva, 2020-11-22:
thx for your answering thats work good
and my last question how can i use `setPixelColor(i, color_from_palette(i, false, PALETTE_SOLID_WRAP, 3));` instead of `setPixelColor(SEGENV.aux1  , SEGCOLOR(0));` in my code?
The purpose of this is to add color selection from the palette
and what does `SEGENV.step` do?

Aircoookie, 2020-11-22:
Try `setPixelColor(SEGENV.aux1, color_from_palette(SEGENV.aux1, true, PALETTE_SOLID_WRAP, 3))`

`SEGENV.step` is just an uint32_t variable availabe to the effect to store a state or something, just like aux0 and aux1. The way it's used differs depending on the effect :)

mehranviva, 2020-11-22:
I tried before, the LED update only applies when I change the color and does not work properly

closed 2021-06-04 by stale[bot]