wled/WLED #1258: Support for high-power LED's

fribse, 2020-10-15:
**Is your feature request related to a problem? Please describe.**
I have set up (so far only one) of these: https://www.aliexpress.com/item/4000003775520.html?spm=a2g0s.9042311.0.0.7ef64c4dU6iqwY
Which is a single LED on 9W, with a big cooling house on it.
But I can't set that much usage in the LED preferences.

**Describe the solution you'd like**
A wider spectrum of values for power usage for each LED.

**Describe alternatives you've considered**
I have so far just set the highest value, and then divided the power supply's capacity to match.

huggy-d1, 2020-10-15:
Maybe this will help: https://wled.discourse.group/t/4-strips-of-led-wont-light-up/960/8

Aircoookie, 2020-10-17:
I agree that the power estimation system was not made with high-power LEDs like these in mind.
I will consider increasing the variable size for the mA/LED :)

blazoncek, 2023-01-25:
Custom mA entry is possible since 0.13

closed 2023-01-25 by blazoncek