wled/WLED #1260: WS2812b don't work anymore

LAR92, 2020-10-16:
Hello I use ws2812b 2m 200 led strip, I flashed wled on the esp8266, d4 on data, gnd on gnd and connected the power for the strip to a 5v 15a external power supply, wled set up everything works fine but suddenly it stopped working, after I set up everything again today it still doesn't work, I reflashed the esp8266 then soldered the jumper wires to the esp to the connector, but it still doesn't work, but the app says that the esp8266 is connected, if i change something in the app nothing happens, do you know what the problem is?
[image]
[image]

huggy-d1, 2020-10-16:
The - lead on the DC supply needs to have a jumper to the GND lead on the ESP8266.
However, be VERY careful because that may put the USB GND lead on the - lead of the power supply.  If it does, it can burn up the USB power source.

I _strongly_ recommend powering the esp8266 with the same 5V supply used on the LED strip and never connect both USB and 5V supply at the same time.

LAR92, 2020-10-16:
Ok, the esp is already connected to the gnd of the power supply and is powered by usb, but i will try it out tomorrow. So positive to VIN and negative to GND? Can the esp 5v and 15a withstand?

huggy-d1, 2020-10-16:
On the NodeMCU in the posted images, you can read printed right on the white background, Vin +5V recommended.

LAR92, 2020-10-17:
Yes, so do I have to connect positive to VIN and negative to GND?  And the ESP8266 can withstand 15A?  I'm sorry for asking so much, I'm just an insecure beginner ...

Aircoookie, 2020-10-17:
The ESP8266 will only pull as many amps as it needs. As long as it's 5V, you're all set!  
That's correct, VIN to 5v and GND to GND. That should work well!  
Keep in mind that if you need to connect the USB port of the ESP (for firmware flashing, etc.) you need to disconnect the 5v wire between ESP and power supply as it could otherwise destroy the ESP.

Hope you can get it to work :)

LAR92, 2020-10-17:
Thanks, I'll give it a try, I'll tell you how it went.

LAR92, 2020-10-17:
So, I connected the esp via VIN and GND to the power supply, when I plugged the power supply into the socket, nothing happened, so the blue led on the esp does not light up and the led strip also does not, so the esp apparently has No power, but if you connect the esp without the cable to the usb then it lights up, then I checked the lines with a multimeter, the positive and negative connections should actually reach the esp and the strip, and the data line reaches the strip apparently not. Did I wire or do something wrong or what do you think is the problem?
[image]
[image]
[image]

sansillusion, 2020-10-17:
You may want to try using the ground pin on the opposite side of VIN. With
my board if I try to power using these two pins I get a bootloop. But when
I used the ground pin on the other side of the board it worked.

LAR92, 2020-10-17:
So I connected ground to the other side, still nothing happened, I wasn't sure what the problem was, so I soldered GND and VIN to a USB cable using a cable, plugged it into a power bank and the esp worked, but why it then? Is the power supply broken? Can I test the strip without a power supply unit by limiting the number of LEDs? Or what ideas do you have?
[image]

LAR92, 2020-10-18:
Can someone help me please?

LAR92, 2020-10-18:
So I once connected the esp and the led strip to another 5v source and lo and behold everything works, before switching on I of course limited the number of leds so as not to destroy my charging station, so it was really my power supply which no longer worked, i will contact the seller and request a replacement, thanks for the help and if i have any further questions i will get in touch.
[image]

closed 2020-10-18 by LAR92