wled/WLED #1261: Logs from WLED

Cuacco, 2020-10-16:
Hi Guys

do you know if there some way to obtain logs from WLED during this is running, it will be greate know if there some issue, problem or wrong behaivour.

maybe you know some way to obtain those logs

huggy-d1, 2020-10-16:
Custom compile where the TX/RX pins of the ESP are not used for anything else, and enable output to the serial (TX) line.
Then, you can monitor the TX line with something like a PI or PC with the right hardware.

Someone posted how they had done this either as an issue or on the https://wled.discourse.group forums.  I do not recall how they did it, but if it were me, I would search and find that post and try to replicate it rather than reinvent (and possible burn up a USB port or UART).

pfeerick, 2020-11-06:
If it's a NodeMCU or Wemos D1 mini or other device with it's own USB to Serial port, you can probably just plug it into your computer. It looks like serial debugging is set at compile time, so you'll need to compile your own firmware. There is probably a better way to enable it, but you'll also need to add `#define WLED_DEBUG` to the top of `wled.cpp` to enable the serial debugging code.

closed 2021-06-04 by stale[bot]