wled/WLED #1262: Toggle effects

ToXIc-Dev, 2020-10-16:
**Is your feature request related to a problem? Please describe.**
I think it would be good to be able to individually disable each effect if you don't want some to be shown

**Describe the solution you'd like**
In the settings you could toggle each effect on or off

**Describe alternatives you've considered**
The only other way I know of doing this is to remove them from the code

Legsmaniac, 2020-10-16:
I'm confused.
Surely the way to "disable" an effect is to not select it in the first place?

ToXIc-Dev, 2020-10-16:
> I'm confused.
> Surely the way to "disable" an effect is to not select it in the first place?

I mean to hide it from the list, the effects list is huge (not complaining, its great), not everyone wants to scroll through that many effects to get to the one they want, so disabling them means hide them from the list.

mike2nl, 2020-10-17:
-> So you'd like a kind of "favourites" list,

That's one of the missing things. Looking at @Aircoookie , haha.

Aircoookie, 2020-10-17:
Hi!
Yes, I too believe the upcoming possibility to save around 100 of your favorite effect configurations will be a good solution to the effect list being a bit long!

Also on my To Do list is to add extra controls besides Speed/Intensity to combine some effects, reducing the overall count.
For example, Candle and Multi Candle are the same except that with Multi Candle, every LED flickers independently. That could be consolidated to a single effect with a `Multi` checkbox. Similarily, the Halloween and Christmas effects are just Running 2 with Orange/Purple and Red/Green colors hardcoded.

Joshfindit, 2020-11-18:
Quick question: What's the best way to pull the effects right out of the code?
I'm running warm white only, so only want a couple.

Aircoookie, 2020-11-18:
@Joshfindit it's not that simple since it wasn't made for dynamically removing effects. You'd need to delete all references to the effect in `FX.cpp` and `FX.h` and modify the total effect count and the IDs of the effects you do want to keep accordingly.

What I'd recommend is setting up a preset for each effect you really like and ignoring the rest of them :)
In 0.11.0 releasing this sunday, you can store a lot more presets and you can even name them to keep track of which preset contains which effect!

Since you are not the only one who wants to hide some effects entirely (it would also have further benefits, e.g. being able to hide all effects from the list that could lead to photoelepsy), I will look into that. It would be enough to hide them after all, any disabled effect would still be compiled in and accessible via the API :)

Joshfindit, 2020-11-18:
> @Joshfindit it's not that simple since it wasn't made for dynamically removing effects. You'd need to delete all references to the effect in `FX.cpp` and `FX.h` and modify the total effect count and the IDs of the effects you do want to keep accordingly.

I was worried it might be something like that.
Thankfully I can live with just the first couple being the only ones, so I just erased all the names that follow those, and they are now unselectable from the UI and HomeAssistant.

> What I'd recommend is setting up a preset for each effect you really like and ignoring the rest of them :)

That works quite well actually, even after my cheat. Thank you ☺️ 

> In 0.11.0 releasing this sunday, you can store a lot more presets and you can even name them to keep track of which preset contains which effect!

!! 🎉!!

> Since you are not the only one who wants to hide some effects entirely (it would also have further benefits, e.g. being able to hide all effects from the list that could lead to photoelepsy), I will [...]

With the cheat I did above, I suspect a shortcut would to track the ID (aka array position) for each name, then selectively remove them from the names array when they are to be hidden

blazoncek, 2023-04-06:
@JensMKofoed 2D effects are hidden by default if you properly configured WLED for 1D (default setting) and have most recent code available.

blazoncek, 2023-05-26:
Instead of hiding WLED offers quick filters in 0.14 and a search bar since 0.12.
I know it is not the same as hiding but is next best thing.

CountParadox, 2023-10-31:
Being able to toggle some effects as disabled would be useful in my use case,

I use home assistant as my control platform, and it's native effects selector uses the wled list, not presets. 

The led on my desk for notifications, has a single pixel... So not many effects work well with it :)

I can shorten and upload the video if requested...

pollinolas, 2023-12-01:
I also use home assistant to randomly switch between effects, but the API that collects the effects makes a total list without discriminating the 2D ones, which leads to the lights remaining fixed without any type of effect.
Does anyone know of an option to solve this, without having to make a manual list of effects?
Greetings and thank you

blazoncek, 2024-03-24:
For anyone interested: https://jeffmission.com/wledcss/

Joshfindit, 2024-03-24:
@blazoncek and anyone else working on this:

I think it would be great to have tags/categories like:

- [ ] Respects chosen colour palette
- [ ] Smooth transition between LEDs
- [ ] X bit (specifically for the support of new LEDs with higher bit depth, but might be helpful for lower-bit effects on ESP8266s)
- [ ] Others in future

blazoncek, 2024-03-24:
I'm afraid that may be an overkill.

Still, WLED supports effect metadata where most of the capabilities and default values are specified. It is accesible via `/json/fxdata` endpoint.
HA or any other software may use that to show or hide certain effects depending on capabilities specified in `/json/info'.

Joshfindit, 2024-03-25:
@blazoncek Sorry, I incorrectly assumed that you were the person behind https://jeffmission.com/wledcss/ and my looping you in on that was to suggest adding those as selections.