wled/WLED #1263: Quick Question re: Functions for Setting State of All Segments

PeteBurgess, 2020-10-16:
Hi folks

I'm trying to create a user mod and I'm not a programmer by any stretch. I can handle a few call(parameters go here) type things but beyond that I'm a little out of my depth. What I'm trying to do is set all LEDs on all segments to white with a colour temperature of my choice (i.e. 4000K) and remove any effects. I was rather hoping the below would do the trick but obviously not. Can someone point me in the right direction, please ?

`         colorKtoRGB(4000, col);
          strip.setMode(strip.getMainSegmentId(), FX_MODE_STATIC);
          strip.setColor(0, col[0], col[1], col[2], col[3]);
          strip.setColor(1, col[0], col[1], col[2], col[3]);
          strip.setColor(2, col[0], col[1], col[2], col[3]);
          effectPalette = 0;
          setAllLeds();`

huggy-d1, 2020-10-16:
getMainSegmentId only gets segment #1.  You want to either override all the segment logic, or update all the segments with all the same effect / brightness / color.  Or, delete all segments except the first one before calling your 1 segment setup.

Aircoookie, 2020-10-17:
Maybe this will work (untested)

```cpp
colorKtoRGB(4000, col);

for (byte s = 0; s < strip.getMaxSegments()) {
  WS2812FX::Segment& seg = strip.getSegment(s);
  seg.mode = FX_MODE_STATIC;
  seg.colors[0] = ((col[3] << 24) | ((col[0]&0xFF) << 16) | ((col[1]&0xFF) << 8) | ((col[2]&0xFF)));
}

setValuesFromMainSeg();
colorUpdated(NOTIFIER_CALL_MODE_NO_NOTIFY);
```

PeteBurgess, 2020-10-26:
That'll do nicely. Thanks for the assist.

closed 2020-10-26 by Aircoookie