wled/WLED #1265: WLED zusätzlich über Schalter an/aus schalten?

eldoman, 2020-10-17:
Hallo zusammen, 
ist es möglich über einen Schalter der direkt an dem Board (z.B. D1mini) angeschlossen ist die LEDs an oder aus zu schalten, ohne dies über die WLAN Verbindung machen zu müssen?

Also nur Licht ein und aus.

Vielen Dank und Gruß.

Aircoookie, 2020-10-17:
Hi,
ja, das ist über einen Taster (Schließer) zwischen GPIO0 und GND ohne Weiteres möglich. An der Unterstützung von Kippschaltern wird noch gearbeitet 👍 
Viele Grüße!

NeariX67, 2020-10-21:
Wenn du einen ESP32 hast, geht das auch mit einem Touch pin. Habe dafür ein Usermod geschrieben, das du in /usermods/ESP32_TouchBrightnessControl findest. In der Readme steht auch wie du es in WLED einbindest (Selbst kompilieren notwendig!)
Ich empfehle ein Stück Alufolie oder irgendwas anderes elektrisch Leitendes an den Pin bzw dann das Kabel anzuschließen, damit man nicht ein winziges Kabel antippen muss.

Camel2k, 2021-01-30:
welchen PIN am ESP benutzt du für den Taster?
Den GPIO 0 gibt es auf meinem Board nicht.

NeariX67, 2021-02-01:
Der Pin ist hier defeniert: https://github.com/Aircoookie/WLED/blob/master/wled00/NpbWrapper.h (Zeile 21)

Camel2k, 2021-02-01:
Hallo, mein Esp32 dev board hat keinen GPIO 0 Anschluß. Oder meinst Du ich sollte den Soucecode anpassen?

NeariX67, 2021-02-01:
Ich denke es wäre besser den Pin anzupassen. Wie im Kommentar steht, sollte es ein Pull-Up GPIO Anschluss sein. Rufe dazu ambesten die Dokumentation deines ESP32 auf, um herauszufinden, welche GPIOs einen Pull-Up Widerstand haben.

closed 2021-06-09 by stale[bot]