wled/WLED #1271: Percentage Brightness for E1.31

CaffeinatedFunctionality, 2020-10-20:
**Is your feature request related to a problem? Please describe.**
I have a few props with a few WLED controllers in a single prop, however, some have different amounts of leds in them. I have an issue with one where I can't get the top and bottom brightness of my matrix correct. The only brightness control that I know works with E1.31 is the brightness limiter, but it's based on a power supply number. If they are using the same 300W power supply I prefer not to use this and just change the brightness with a percentage.

**Describe the solution you'd like**
An option to use the brightness limiter on a percentage level and not just a power supply amount.

**Describe alternatives you've considered**
I've tried playing with the brightness limiter current but can't seem to get the props perfect this way. Especially since each prop is different. My show currently gets very different levels of brightness and I just want them all to be the same brightness.

Aircoookie, 2020-10-20:
Hi!
There is a `Brightness factor` option in LED settings which sounds exactly like what you need :)

CaffeinatedFunctionality, 2020-10-20:
There is but it doesn't work with E1.31. I'm using multicast and have to use force max brightness so I don't have to turn on my WLEDs before the show. 

⁣Get BlueMail for Android ​

CaffeinatedFunctionality, 2020-10-22:
I can confirm that does not work with force max brightness since that gets overridden. I use force max brightness because I don't want to have to turn on and off the WLED sequences during the show. Is it possible we could add an option like the brightness limiter, but as a percentage that won't be overrun on force max brightness? I probably could do this myself and submit a PR if that helps. Thank you!

Aircoookie, 2020-10-22:
Oh, that is an oversight. The brightness factor should be honored in the "max brightness" that is set by the force brightness option, but that indeed ignores the factor and always sets to 255. Will fix tonight :)

Aircoookie, 2020-10-29:
"Tonight" in Aircoookie time is a week later 😅 anyways, should be working now with the latest source!

closed 2020-10-29 by Aircoookie

CaffeinatedFunctionality, 2020-10-29:
You are awesome. I fixed it locally but I'm sure you did a better job I will check it out. 

⁣Get BlueMail for Android ​

commit 373d12be (2020-10-29): Realtime max. brightness now honors brightness factor (fixes #1271)

- Colorful effect now supports palettes
- Added C9 2 palette (#1291)
- Improved C9 palette brightness by 12%
- Disable onboard LED if LEDs are off (PR #1245)
- Added optional status LED (PR #1264)
- Realtime max. brightness now honors brightness factor (fixes #1271)
- Updated ArduinoJSON to 6.17.0