wled/WLED #1272: Progress Bar/Timer Effect

tylerwmarrs, 2020-10-20:
**Is your feature request related to a problem? Please describe.**
There is not a problem, I simply want to suggest a feature.

**Describe the solution you'd like**
Allow users to specify a timer, in seconds, that the led then slowly fills based on time. Optionally, we can have the color specified, flash N times when complete option, and time before shut off with color option. I have written my version of this within ESPHome + FastLed.

Please excuse my potentially poor implementation shown below as I am still new. I'm just investigating ESPHome + FastLed and WLED at this time.

```YAML
effects:
      - addressable_lambda:
          name: "30 Second Timer"
          update_interval: 0ms
          lambda: |-
            static int index = 0;
            auto turn_off = id(livingroomled).turn_off();
            int num_flashes = 5;
            static bool flash;
            static int flash_count;
            int now = millis();
            static int last_run = 0;
            int diff;
            static int diff_sum;
            int ms_delay = 500;
            int off_delay = 5000; // 5 seconds
            int flash_delay = 500;
            int brightness = 125;
            int constant_brightness = 175;
            
            if (initial_run) {
              flash = true;
              flash_count = 0;
              last_run = 0;
              index = 0;
              diff_sum = 0;
              flash = true;
              
              // clear leds
              for (int i = 0; i < it.size(); i++) {
                it[i] = ESPColor(0, 0, 0);
              }
            }
            
            diff = now - last_run;
            diff_sum += diff;
            
            if (index < it.size() && diff_sum >= ms_delay) {
              it[index] = ESPColor(0, brightness, 0);
              diff_sum = 0;
              index++;
            }
            
            if (index >= it.size() && diff_sum >= flash_delay && flash_count < num_flashes) {
              if (flash) {
                for (int i = 0; i < it.size(); i++) {
                  it[i] = ESPColor(0, brightness, 0);
                }
                flash = false;
                flash_count++;
              } else {
                for (int i = 0; i < it.size(); i++) {
                  it[i] = ESPColor(0, 0, 0);
                }
                flash = true;
              }
              diff_sum = 0;
            }
            
            if (flash_count >= 5) {
              for (int i = 0; i < it.size(); i++) {
                it[i] = ESPColor(0, constant_brightness, 0);
              }
            }
            
            if (diff_sum >= off_delay) {
              turn_off.perform();
            }
            
            last_run = now;
```

**Describe alternatives you've considered**
I have implemented it in FastLed on ESPHome as shown above.

**Additional context**
It would be great to expose the options within Home Assistant. I'm currently thinking of how to do this with my current FastLed implementation instead of having multiple effects for different time intervals.

Thank you for your ideas for making WLED better!

NeariX67, 2020-10-21:
The main problem I see with this, is that the slider from the webinterface only has 256 values. So it would be very hard to take a precise interval like 1 minute or something. Which leads to another question: What time interval do you set for the animation? 0-10 seconds? 0-10 minutes? 0-10 years? 0-10 lightyears?

tylerwmarrs, 2020-10-21:
@NeariX67 is adding a simple integer input difficult to do in the current web interface? In regards to the time interval, we can simply compute the animation rate by taking the number of leds and the timer integer value.

For example, assuming the input value via the web UI is in seconds with a value of 300 (5 minutes) and 60 leds:

1. Convert seconds to milliseconds
```
millis = seconds * 1000
millis = 300 * 1000 = 300000
```

2. Compute the animation_rate (light up rate) of each pixel
```
animation_rate = millis / num_leds
animation_rate = 300000 / 60 = 5000
```

In this example, each pixel would be lit up every 5 seconds. Does this help to clarify the time interval animation?

blazoncek, 2022-07-25:
Similar effect exists. It is called Percent and uses values 0-100 to light up the amount of LEDs corresponding to the set percentage.
Values from 101-200 will light up LEDs in reverse order.
Closing as this should fulfil majority cases where such behaviour is needed.

closed 2022-07-25 by blazoncek