wled/WLED #1275: Newton's Crable Effect Feature Request

robisoel, 2020-10-20:
**Is your feature request related to a problem? Please describe.**
It would be nice to have an effect that worked like a Newton's Cradle.  Perhaps something like shown here:  [https://www.youtube.com/watch?v=Vuwqrug2fx8](https://www.youtube.com/watch?v=Vuwqrug2fx8)

**Describe the solution you'd like**
One color mode would be simple
Two color mode would change the color of the ball(LED) that "swings" out until it comes back
Three color would display a different background color.  

**Describe alternatives you've considered**
None at this time

**Additional context**
See video:  [https://www.youtube.com/watch?v=Vuwqrug2fx8](https://www.youtube.com/watch?v=Vuwqrug2fx8)

Thank you for your ideas for making WLED better!
Thank YOU for making such a cool LED software tool!  I love it.

FilipParyz, 2020-10-21:
I'm new to this project but I'd love to contribute to it! I'll try to make something that solves your problem 👍

robisoel, 2020-10-21:
Thanks @FilipParyz !