wled/WLED #1278: Add MicroSD Support AND Falcon Pi Player Remote Support

MrPaulAR, 2020-10-21:
**Is your feature request related to a problem? Please describe.**
Wifi is inherently noisy and congested which often results in choppy e131 & DDP playback.  Since ethernet isn't an option for most esp8266/esp32 devices we must work with what's been given.  

**Describe the solution you'd like**
Add support for a MicroSD card, reading FSEQ files (maybe compressed if the CPU can handle it), and FPP/xSchedule remote support.  

**Describe alternatives you've considered**
N/A

**Additional context**
QuinLED doesn't support MicroSD today but other esp "Capes" do such as the ESPixelStick hardware.  The ESPixelStick software also has the required remote support built in and sd card support which WLED can run on.  Perhaps if this feature were to exist then Quindor may even be more receptive for adding a v2 of his boards with MicroSD support.  Especially on the Quad boards.

I realize this is a major ask and I'm hopeful you can understand where this is coming from.  With more and more data coming from xLights this will be a significant improvement.

pbolduc, 2020-11-14:
Have you seen https://github.com/ShaunPrice/FSEQLib. This can read fseq from SD card and display them. Doesn't have a pretty UI like WLED.

I would have to think how this could be integrated. I noticed the some what incompatible licenses between WLED (MIT) and FSEQ (GNU).  

1. Could FSEQ be treated like special effect? Copies the next frame from the file.
2. What config options required? File name? Universe number?

MrPaulAR, 2020-11-14:
Don't see why the FSEQ couldn't be treated as a special effect.  We do something like that today in FPP & xScheduele.

The FSEQ format is detailed in the [fseq spec](https://github.com/Cryptkeeper/fseq-file-format).  Some assumptions would need to be made by WLED regarding the channel numbers.  Now that DDP is an option ya could just use the absolute channel numbers.

pbolduc, 2020-12-19:
I do own a couple of micro sd shields so I can can help test once this gets support.

blazoncek, 2022-10-10:
#2292 and #2413 may add SD card functionality.