wled/WLED #1279: Variable brightness based on time of day

hoppertje, 2020-10-22:
First of all thanks a lot for creating WLED!

**Is your feature request related to a problem? Please describe.**
Not a problem, a feature. When turning the lights on during the day I would like to have the leds on full brightness while after the sun has set down only half brightness is needed. 

**Describe the solution you'd like**
A solution to this  would be to have the option to change the default brightness based on the time day of day. The checkbox for default brightness after power up would then look like this: 
Time              | Brightness 
00:00 - 08:00 | 100
08:00 - 19:00 | 255
19:00 - 00:00 | 100

**Describe alternatives you've considered**
While its possible to set the brightness with timed macros. This information is lost on a power up/reset. In addition to this my setup has the NodeMCU connected to the led power supply which is switched off with a smart outlet. 

Thanks in advance!

huggy-d1, 2020-10-22:
This could be implemented now with a usermod.  The usermod would need to store the time-of-day start/stop/brightness levels for the, in your specific case, 2 ranges.  It's either between 800 and 1900, or it is not.  If it is within the range, 100% brightness.  Otherwise, 100/255*100=39%  brightness.  These could be established either via JSON or HTTP API calls, json preferred in v2 usermods, or additional UI created for the usermod.

Essentially, the usermod would force NTP enabled in when usermod's setup() is called, and in loop() it would adjust brightness for every segment to either 100% or 39% depending on time-of-day relative to the required times.

The only times that matter are fullBrightnessStart and fullBrightnessStop.  If the current time is not between those times, 39% brightness is set.

I imagine someone has already done something very similar and posted how they did it on this wled github in a (now) closed issue here.

hgregorian, 2020-10-23:
I know this may not be necessarily be the standalone answer I think is being requested, but I recently implemented this via home-assistant automation.  I'm using sun elevation to make the brightness changes so that it best matches the outdoor brightness, where a time-based solution can obviously vary.  In this way, brightness level changes both as the sun rises and as the sun sets without duplication of code.

huggy-d1, 2020-10-23:
If you want to go a bit further, add a LDR (light dependent resistor) and customize it dynamically.  That way, it can automatically adjust intensity for any situation, such as during an eclipse, full moon, cloudy day, overcast day, rainy day, snow day, and so one.