wled/WLED #1281: WLED for SP501E

DarKOrange75, 2020-10-24:
Hi,

Thank you for this great software, I was able to flash a SP501E with a bin someone was kind to compile for me.
But now I am stuck to this version.
My question is would it be possible to provide also a version for this controller so people with SP501E would be able to flash via OTA and stay up to date?

Thank you :-)

Aircoookie, 2020-10-26:
Hi!
The problem with OTA on the SP501E is that it only has 1MB of flash, and WLED's code size has grown too large for it to fit twice (which is required for the OTA update to work). We got #1266 though which might make OTA on 1MB boards work for a few more versions. (although I am in the process of adding filesystem support, which will use a lot of additional space).

So I can't say for sure yet whether it is possible, a single megabyte is very tight, but I want to support the SP501E as well as the Sonoff units as well as possible. These are also the hardest to flash wired since they don't have an USB port like the 4mb dev boards.
Will also add a platformio environment for the SP501e, that will make it easier for me and others to provide updated binaries for it :)

Operation760, 2020-10-26:
AirCookie- you and the community support is off the charts. Glad you're committed to these products! 

Hi DarkOrange- I originally reversed engineered the 501e. I'm planning to move my builds and use cases to the latest rev in the next few weeks as I have multiple light projects I'm kicking off. If you're not in a rush, I'll be cranking out a lot of experimental builds and had been planning on seeing if I could get OTA working in the previous release. I deliberately left out OTA after trying and failing a few times. Given the issues then and the fact the code base is only growing, I don't know if I'll be successful and thus you might want to wait for AirCookie. If you're willing, can you share what features you absolutely need (Alexa, MQTT, etc.) and I'll see if I can get at least an bare-bones build produced.

DarKOrange75, 2020-10-26:
Thank you @Aircoookie and @Operation760 ☺️🙏🏼

@Operation760 the most important to me is to be able to update via OTA to get new effects and palettes and also MQTT since I can get Alexa/Google Assistant and Siri through Home Assistant.

I understand it will be more work so thank you very much!

huggy-d1, 2020-10-26:
For SONOFF basic's, at least some of the older ones, this might help make them love WLED w/ file system: http://www.andremiller.net/content/upgrading-sonoff-wireless-smart-switch-flash-memory-esp8266
And over at SuperHouse (Jonathon Oxer): https://www.superhouse.tv/21-six-sonoff-secrets/ he shows several Sonoff tips, one of which is the memory upgrade process.  It may be outdated for newer Sonoffs, especially for those using ESP8285.

chris-guthrie, 2020-11-04:
For OTA updates on the limited storage would it be possible to do something similar to what the Tasmota project does where you install a minimal firmware to free up flash, then allowing another OTA to the newer larger firmware?

Just a thought and thanks for all the work on the project @Aircoookie

huggy-d1, 2020-11-05:
@chris-guthrie You can do that already.  
Instead of loading the WLED update for an OTA update, load up the minimal Tasmota firmware when WLED prompts for the firmware file.
Why should @Aircoookie recreate a minimal OTA firmware when Tasmota already did it?  

Maybe after WLED version 2.0 is released, he might have enough time to devote to something like this.  Maybe someone familiar with Tasmota code could sketch up a proposal and do a pull request.

chris-guthrie, 2020-11-05:
@huggy-d1 If that the case then great I didn't know if that would work or not when it comes to installing the WLED firmware over Tasmota. I will have to try that and see it it works.

Agreed I in no way want to waste anyone time @Aircoookie creating something that's not needed. It was not the intention of my post, was just thinking of other ways that may help support these devices with limited space.

chris-guthrie, 2020-11-16:
@huggy-d1 so I tried this method using the Tasmota min firmware and it never worked for me. But I noticed that @Aircookie had a similar project in Aircoookie/ESP8266MinimalHTTPUpdater. I built that installed and was able to reinstall the image I'd built for my SP501E  devices. This is great as I have a working way to update the controllers without having to resolder 👍 Thought this might be interesting to other asking the same question.

blakadder, 2020-12-12:
my SP501E came with 2mb memory. i compiled an ota supporting version of 0.11

Leo-PL, 2021-01-11:
@Aircoookie I think that you could include SP501E as well in your releases, as support is there, OTA works, and at least new ones come with 2MB Flash. I wonder if, given that Gzip support was merged, is it possible to squeeze the full build in 1MB version.

barkster, 2021-11-12:
Is there a precompiled bin available for the sp501e with 2mb flash? I  installed a different one and when I try to set my led count to 360 it says it's too many for this hardware even though the sp501e says it can handle 1024 leds

Leo-PL, 2021-11-12:
@barkster just use 2MB binary and set GPIO accordingly, it's configurable runtime now.

closed 2023-10-02 by softhack007