wled/WLED #1283: Trying to connect to Amazon Echo Dot.

suhassamarth, 2020-10-25:
Hello @Aircoookie , i'm having a nodeMCU and Analog LED's connected and all is working fine except alexa cannot find the device under philips hue nor other devices. NodeMCU hasn't connected to any UpNP 
[image]

Please help me out.

closed 2020-10-25 by suhassamarth

Aircoookie, 2020-10-25:
Hi, did you manage to find out what the issue was?

suhassamarth, 2020-10-26:
Yes, @Aircoookie , mine is a 3rd Gen Echo Dot. On an unusual occasion without any uPnP client being subscribed I had to restart(Router, Node MCU & Echo Dot 2 time) and change the name of Alexa Invocation name. Then it found the device.

Anyway thanks for the response @Aircoookie . Love the work 👍