wled/WLED #1284: 1 Pixel dead when using xLights

PKCubed, 2020-10-25:
I just installed WLED on my esp8266 and I have a similar, if not exactly the same problem. I’m using XLights to control my LEDs. They all work except for the 1st LED on the 2nd universe. (It could also be the last LED on the first universe)

I also just reinstalled xLights, and that doesn't seem to make a difference.

Aircoookie, 2020-10-25:
#1187 :) It is already fixed in source, so the error will disappear if you compile from source or once I release v0.11.0

closed 2020-10-25 by Aircoookie

PKCubed, 2020-11-16:
Has v0.11.0 been released yet? If not, when will it be?

Btw, thanks for making this. I'm making a Christmas light show for the first time and everything other than this has been working really well.

Aircoookie, 2020-11-16:
No it has not been released yet, but will be for sure this week!

PKCubed, 2020-11-16:
Great, thanks!