wled/WLED #1285: Compiling with GPIO4 for LED and GPIO5 for Button

MNatzke, 2020-10-25:
I tried to change the PIN Settings in NpbWrapper.h like this to compile WLED for the setup mentioned in the subject:
```
//PIN CONFIGURATION
#ifndef LEDPIN
#define LEDPIN 4  //strip pin. Any for ESP32, gpio2 or 3 is recommended for ESP8266 (gpio2/3 are labeled D4/RX on NodeMCU and Wemos)
#endif

#ifndef BTNPIN
#define BTNPIN  5  //button pin. Needs to have pullup (gpio0 recommended)
#endif

#ifndef IR_PIN
#define IR_PIN  2  //infrared pin (-1 to disable)  MagicHome: 4, H801 Wifi: 0
#endif

#ifndef RLYPIN
#define RLYPIN 12  //pin for relay, will be set HIGH if LEDs are on (-1 to disable). Also usable for standby leds, triggers,...
#endif

#ifndef AUXPIN
#define AUXPIN -1  //debug auxiliary output pin (-1 to disable)
#endif

#ifndef RLYMDE
#define RLYMDE  1  //mode for relay, 0: LOW if LEDs are on 1: HIGH if LEDs are on
#endif
```

Everything works "fine" but the system crashes very often when I change the lights on the UI or via Home Assistant.
"fine" means: The Button is switching the light on and of and the led Strip is doing what i want. But crashes very often.

I tried the binary provides in the release section for LED on GPIO4. This version does not crash but the button is not working. Guess it is not assigned on GPIO5.

My question:
Did i missed some configuration where GPIO5 (my button) is used different? Guess there is some kind of a conflict.
Maybe one interesting fact: The System sometimes crashes by entering the Sync-Setup without changing any light settings.

Unsing a NodeMCU

Aircoookie, 2020-10-26:
Hi!
What is your build environment (IDE, settings)?
You can also send the output of `[WLED-IP]/json/info` to give me a clue what could be going wrong :)
GPIO2 really is the most reliable pin as it does not require bitbang, but there shouldn't be frequent crashes when using the other pins.

MNatzke, 2020-10-26:
Thanks for the quick respond. 
I am using platform.io to compile the binaries. Took the sources provided on your release page and changed only the mentioned settings in my initial post.

Thats what /json/info delivers:
`{"ver":"0.10.2","vid":2008310,"leds":{"count":45,"rgbw":false,"wv":false,"pin":[4],"pwr":666,"maxpwr":4000,"maxseg":10,"seglock":false},"str":false,"name":"WLED","udpport":21324,"live":false,"lm":"","lip":"","ws":0,"fxcount":113,"palcount":52,"wifi":{"bssid":"38:10:D5:82:E1:17","rssi":-69,"signal":62,"channel":1},"arch":"esp8266","core":"2_7_4","resetReason":2,"lwip":1,"freeheap":13944,"uptime":17,"opt":247,"brand":"WLED","product":"FOSS","mac":"68c63a993ca4"}`

What i am wandering about is, that the system crashes after navigating to the Sync Option. It is impossible to open or navigate to IP/settings/sync?. The system will crash.

Aircoookie, 2020-10-26:
Ah, I believe I've got it figured out. Platform.io, by default, builds with the `travis_esp8266` environment. That build is not stable and only meant for continuous integration testing. Try commenting out the `default_envs` in platformio.ini and instead select the `d1_mini` environment by uncommenting that. Than it should work :) (the reason it crashes is likely excessive memory usage due to everything being compiled in)

Since you are building from source, I would also give the latest master a try. It contains a few smaller fixes since 0.10.2 but I haven't had time to make a new release yet :)

MNatzke, 2020-10-27:
thank you very much for your support. It was actually due to the compile options. Unfortunately I still observe occasional crashes. Of course not as often as before and also not already while navigating in the UI.
My impression is that complex effects make a crash more likely.
I'm going to install your standard release on a "naked" esp8266 and test it without connectors to see if there are crashes when playing complex effects.
Currently the systems with effects crash after 3-6 hours, but then they run more stable (16h so far).

Tried your current master release as recommended as well as the sources of release 0.10.2

MNatzke, 2020-10-28:
OK, testet some settings and have to say, that the recommended PIN Setup GPIO02 for LED and GPIO0 for the button is much more stable than my setup.

So, took my dremel and cut some tracks on my PCBs and did some re-soldering.

closed 2020-10-28 by MNatzke