wled/WLED #1286: Categorize effects in list

Aircoookie, 2020-10-26:
**Is your feature request related to a problem? Please describe.**
Effect list is too long and may seem overwhelming

**Describe the solution you'd like**
[From SpaceCadets on Discord]
Hey Everyone, I wanted to see how possible it would be to categorize the effects in groups instead of one long list of buttons? I think it would be awesome to have the buttons be a bit smaller, possibly round corner squares, that are arranged in rows of two. How difficult would it be to modify the gui to have drop down groupings of effects that are expanded or hidden with a tap? For example having "Noise Effects/Smooth Effects", Sound Reactive (for the sound reactive version), Intense (strobe, etc)
I think this would make the gui much easier to use, and not seem like there are so many options that it is overwhelming.

[Aircoookie]
 that sounds like a very good idea, since the effect list is indeed getting a bit long, so having a few (but not too many so you still know where to look for a given effect) categories (maybe Calm/Physics/Noise/Intense/Basic/Other) would be great. Of course I would also add a toggle to revert to the alphabetical list for finding a specific effect quickly!

**Describe alternatives you've considered**
Reducing overall effect count

**Additional context**
Should probably be implemented purely on JS side and not change the api for now

SpaceCadetLights, 2020-11-07:
I am working on a modified version of the GUI that organises the effects, just by modifying the ui files in the data folder, and the list of effect names in the source code. I'll show more specifics on a few days once it's complete.

blazoncek, 2022-07-25:
This might be easy to implement by enhancements included in #2610

blazoncek, 2022-10-10:
In 0.14 it is possible to add extra info to effect data opening a possibility to group effects.