wled/WLED #1289: Default Settings beim booten werden nicht gespeichert.

Tihass1975, 2020-10-27:
Hallo an das WLED Team,

wenn man unter **led References** -> **Defaults** -> **Apply Preset at boot** zum Beispiel auf Slot 1 setzt, werden Einstellungen wie  **Default brightness** und **Turn LEDs on after power up/reset** ignoriert. Wenn man **Apply preset 0 at boot** belässt, werden die Einstellungen berücksichtigt.

Ich benutze WLED 0.10.2 in Verbindung mit einem WEMOS D1 mini Pro und APA102 LED Stripes.

Gruß
Matthias

csantelman, 2020-11-30:
English translation - 
If you set, for example, slot 1 under led References -> Defaults -> Apply Preset at boot, settings such as default brightness and turn LEDs on after power up / reset are ignored. If you leave Apply preset 0 at boot, the settings are taken into account.

Aircoookie, 2020-12-01:
Hi! In v0.11.0 habe ich dieses Verhalten geändert, die beiden Einstellungen sollten nun auch bei gesetztem Boot preset funktionieren :)

Hi! I have changed this behavior in v0.11.0, the two settings will now function correctly with boot preset enabled

closed 2020-12-01 by Aircoookie

csantelman, 2020-12-01:
I'm confused - I still can't get this to work.  In my case, I want to use a wifi enabled switch to control the power supply, which would then control the lights.

- Using the wled app on my phone - I set the lights to be a non-default color - blue.
- Navigate to config/LED Preferences, check the "Set Current preset cycle setting as boot default" box
- Save (Settings saved message appears)
- Power off the lights via the phone app, power back on, and the light is blue.  This is expected behavior.
- Turn off the wifi switch, lights go off (expected)
- Turn on the wifi switch, lights come on, but are back to the defualt color (unexpected)

Aircoookie, 2020-12-01:
@csantelman please don't comment the same in multiple issues. Replied to you in #1414 :)

csantelman, 2020-12-01:
@Aircoookie  - sorry...

Aircoookie, 2020-12-01:
no worries :)

tourer4778, 2022-12-06:
Hello,
Is it also possible that the LEDs are off, after power on (relay too) ?