wled/WLED #1291: Standard colored Christmas lights palette

rjmcfadd, 2020-10-27:
Can you add a palette that has the 5 standard colors of Christmas lights (Red, Blue, Green, Orange, and Yellow)? Currently, you can only select 3 colors unless I'm doing something wrong. Thank you!

Aircoookie, 2020-10-28:
Hi!
We have the "Colorful" effect, which (make sure intensity is set in the upper half) has 4 colors, Red, Blue, Green and Yellow and we have the C9 palette which resembles Christmas lights (Red, Amber, Green and Blue) and looks very good with the e.g. Colortwinkles effect. But I might also add a "C9 2" palette that also includes yellow :)

rjmcfadd, 2020-10-28:
Hello, thank you for your reply! I missed these I guess because there are so many great options. That would be great if you could add the "C9 2" palette. Thank you!

closed 2020-11-20 by Aircoookie