wled/WLED #1293: WLED working from macbook browser, but not android/iphone app or browser

bbrandal, 2020-10-28:
I have a really strange problem. I have a D1 mini (ESP8266, this specific version https://www.amazon.com/gp/product/B081PX9YFV/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1 ) programmed with WLED version 0.10.2 (WLED_0.10.2_ESP8266.bin).

From my macbook everything works, I can connect and control the WLED from Firefox, chrome and Safari browsers. But from mobile browser and WLED app it cannot connect.

During **configuration** it also works from the phone, and I can connect to the WLED AP using my mobile (Xiaomi note pro 10, Chrome browser) and set the WLED up to connect to my home network. But then when the WLED is connected to my home network the phone cannot, neither browsers nor app connect.

I have checked all the obvious pitfalls, like both phones, the macbook and the WLED are on same wifi (and no cabled network on my macbook, so definitely connected over wifi). Also the phone has no issue connecting to WLED AP during the setup phase and configure the WLED. Only after its on my home wifi the phone (browser and app) can no longer connect.

I also tried connecting to WLED from my home assistant. The WLED is not auto discovered (I have mDNS enabled) and integrating manually using the WLED integration also reports "Failed to connect to WLED device.", but at same time my macbook happily talks to and controls WLED.

This behaviour is really strange. Could there be a bug in 0.10.2 where for whatever reason browser on macbook can connect, but mobile browsers/app and home assistant cannot connect? Everything else works as it should on all devices and on home assistant.

jes1417, 2020-10-28:
I had this problem on Android, don't know what caused it but I ended up clearing cache in the WLED app and everything on mobile worked again. I couldn't even ping the board from my phone when this happened it was very strange.

bbrandal, 2020-10-28:
Thanks for the suggestion. I tried both clearing app cahce and all app data, but still same problem. I have now also tried a different ESP8266 (NodeMCU board) and an ESP32-WROOM-32 based module (with the WLED_0.10.2_ESP32.bin release). Same behavior. All good from macbook browser, but still no connection from phone app or browser, or from Home Assistant, except connecting to AP and configuring WiFi works fine from phone browser

Aircoookie, 2020-10-28:
Hi and sorry for the issues!
Are you connecting to the WLED IP or to the .local mDNS address? For android, only the former is supported.
The mobile apps should work in any case though. My only suspicion is some kind of firewall that either prevents mDNS or HTTP traffic to work.
It could be that the module does not connect to your WiFi in the first place, but then it shouldn't be reachable with the macbook either (unless that somehow autoconnected to WLED-AP again)

bbrandal, 2020-10-28:
Hi and thanks for quick reply.
I am connecting to the WLED IP. My WiFi router is an ubiquiti UDM (UniFy Dream Machine), and I have also turned of the auto optimization that could drop multicast incl mDNS. I also used "Discovery" app and verified that the WLED DNS is broadcasted, although should not matter when I use the IP. 
[image]

I have also verified that there are no firewall rules set up.

And yes, I have verified every time that both the macbook and the phone are indeed on the same network as the WLED, but still only macbook can access it. Ill try to dig further, maybe replace with another router to rule anything with the UDM out...

bbrandal, 2020-10-28:
I found the problem. In the UDM under the WLAN options there is a setting "Enable minimum data rate control" which was by default set to 6Mbps. There was an info box saying this could affect compatibility, and sure thing, turning this off solved the problem and I can connect to the WLED from the mobile as well, including using auto discovery.

Thanks again @Aircoookie for the prompt reply and for an absolutely fantastic piece of software and firmware. Hopefully I can somehow contribute in the future, and if not for anything else, maybe someone with similar issue can find this report useful.

Aircoookie, 2020-10-28:
Awesome that you found out what the problem was! Strange that there is such a setting in the first place (probably to make  slow devices not block the channel for too long, but in turn disabling them almost completely, which doesn't make sense?)

I'm very glad you like WLED so much :)

closed 2020-10-28 by Aircoookie

Prowling123, 2021-06-21:
Hmm, i also have the unifi network. However, data rate control is off, but I still cannot reach the esp32. It works when I reboot it. I tried it on the Iphone and a huawei device. Both do not open the page (unavailable). The laptop however (chrome) works fine and from my hubitat hub it also works. 
from my android phone somehow I am not able to ping the esp. Does anyone know what this could be?