wled/WLED #1294: emi causes wled to switch between effects

hamster65, 2020-10-28:
A radio amateur in our house causes wled to cycle between solid color and one of the effects when he is transmitting :)
I suspect it is the infrared input pin and it seems it cannot be deactivated. Is the pullup for this pin active or is it floating by default? Are there other inputs that could cause this?
No other devices in our house are or have been affected, so it could be a floating pin.

pille, 2020-10-31:
according to https://github.com/crankyoldgit/IRremoteESP8266/blob/b77b0e1ae600d48d7352042b538a53c7e0405564/src/IRrecv.cpp#L228 the PIN is not pulled up by default. you may change it by changing `    irrecv->enableIRIn();` to `    irrecv->enableIRIn(true);` in https://github.com/Aircoookie/WLED/blob/2049e8b8df1d9c39f892b207fbfba543cea6a284/wled00/ir.cpp#L492.

you can also disable IR completely by compiling with `-D WLED_DISABLE_INFRARED`

@hamster65: can you verify this?

hamster65, 2020-10-31:
Thanks pille! I connected a 10k pullup to the IR pin and had no more issues with emi afterwards but i will know for sure in two days.

For me this is solved but in general, inputs should always have a defined state, they should never be floating.  Disabling IR support with a compiler option does not help users who use the binaries. I can think of the following clean solutions:

- (Disable IR in the binaries, enable by compiler options)
- (Always connect an external pullup (add this to the schematics) )
- (Always enable pullup -> might be incompatible with some IR sensors)
- Option to disable IR
- Option to enable (default) / disable pullup, with annotation "set to enable if IR input not connected"

The last one would be the most flexible, I guess.

I am not sure if enabling the internal pullup would work with every IR sensor. To keep wled as compatible as possible, I would either add an option to enable / disable the internal pullup (set to enable if IR sensor is not connected, should be default), or I'd add an option to disable IR.

pille, 2020-10-31:
i agree, this should be fixed.
but to clarify: my request was just to confirm your thesis.
can you please verify that the code change would work, so we can do a PR?

Aircoookie, 2020-10-31:
Good to know that the floating IR pin was causing the issue!
@pille's suggestion of using the pullup mode is a good one. I doubt it would cause issues with IR sensors as they typically have pullups themselves, but that remains to be seen.

The IR can actually be disabled at runtime. If the `Infrared Remote` setting in Sync settings is set to disabled (which should be the default), any received signal should be ignored. @hamster65 what is that dropdown set to for you? If this happened in the disabled state, something weird is going on.

hamster65, 2020-11-01:
Infrared remote is in fact set to disabled ! Hm.
On/Off button is enabled but it has the pullup enabled and it does not explain the switching between effects. Are there any other inputs?

hamster65, 2020-11-01:
Update: My neighbour was on air again. This has nothing to do with the IR input, the LEDs went crazy again, even with the added resistor. Last time, I could see a pattern in the LED, like the circus effect, but this was misleading, it is random flickering mixed with the correct data from WLED.

I touched all pins, no change. Then I set brightness to a very low level. Still, the LEDs flicker brightly. The peek function does not reflect what the LEDs are doing, so the real issue must be the data line of the WS2812B strips and it is not something WLED sends to the LEDs. When I switch it off in the web interface, the flickering stops. I guess WLED stops sending data in this case?

I have got another WLED on a breadboard with a short WS2812B strip attached. It also flickers... perfect test candidate. I'll connect the oscilloscope next Wednesday.

pille, 2020-11-01:
do you use a resistor on the data pin?

hamster65, 2020-11-01:
no

Legsmaniac, 2020-11-02:
I would imagine that the Ham radio transmission is being picked up either by the Arduino device itself in general or is being received by the data wire, upsetting the data generated by the Arduino.  
I would suggest two things to try, one or the other, or even both......

Can you perhaps try placing the Arduino in a metal box (ensuring nothing will be shorted out on the Arduino obviously) and ground the box, thereby producing a Faraday Cage to protect the Arduino from the radio signals?

Can you try replacing the (I presume) single data wire from the Arduino to the LED strip with a coax cable, grounding the shield/screen wire?

Hopefully, either or both of these modifications should help.
Good luck.

hamster65, 2020-11-02:
The Arduino is an ESP12 with a metal can shielding. I think it is connected to GND, but the power supply is not mains referenced, so GND is not connected to PE / earth. I'll try that. Next HAM transmission is on Wednesday :)

I was able to test with the scope yesterday. When the HAM is transmitting, there is a small ripple on the data line, starting just before the middle of the screen.
[image]

Looking at the WS2812B datasheet, logic high should be at least 0.7xVCC. That would make:
min. 3.5V at 5V VCC. Should not work with the 3.3 V ESPs at all, but does :)
min  3.4 V for the "big" lamp at 4.9V.
min. 2.7V at 3.94 V for the small led strip (drop due to usb cable)

I'd say: For the big lamp, EMI might cause bit flips, the scope shows I am close to the minimum for logic high. But it should not be an issue for with the small lamp. I also checked for spikes etc on the power rail and it looked ok, but the HAM was not transmitting any more at this time.

I have a level shifter board somewhere that I want to try with the small lamp to get a full 5V logic signal.
Reducing the supply voltage to around 4.7V would be another interesting thing to try but the supply has no adjustment.

harueg, 2020-11-03:
Hi @hamster65 
Do you what frequency your neighbor is sending ?
It is good practice to shift the level from 3.3V to 5V. either some HCT or an adum1200.

hamster65, 2020-11-09:
The level shifter solved that problem. The lamp without shifter is still going crazy, but the one with shifter worked without any flickering.
I used one of those 4x bi-directional shifter boards that you can get from ebay. Its a mosfet and two resistors for each channel.

I think my neighbour is transmitting somewhere around 50 Mhz.

blazoncek, 2022-10-10:
EMI is almost impossible (or infeasible) to mitigate in software.
IR can be disabled in settings pages (though IR is not strictly speaking EMI).
Proper shielding of WLED device and external antenna are essential in fighting EMI.
Closing.

closed 2022-10-10 by blazoncek