wled/WLED #1295: LED arangement / segments lost after power cycle / how to configure for fire 2012 effect?

hamster65, 2020-10-28:
I have 
36 leds from bottom to top,
36 leds from top to bottom and another
36 leds from bottom to top.

I created segments (the second one inverted) but once wled is restarted, they are gone.
I also could not find a good setting for the fire 2012 effect where it uses all LEDs and "burns" from the bottom.

Legsmaniac, 2020-10-28:
This piqued my curiosity, I had to try it.  
I was able to get the Fire 2012 Effect all the way to the top (all LEDs) from the bottom by increasing the Intensity slider and adjusting the Speed slider to suit.  Did you try this?
Yes, restart WLED and your segment settings are reset.  However, if you set up your segments and then save Slot 16, all your segments will be saved (only Slot 16 does this - at present)  Then you can set it up so that Slot 16 is the default when WLED starts (see settings).
Hope this helps.

hamster65, 2020-10-28:
Thanks, I'll try this...

hamster65, 2020-10-31:
Worked, thanks!

closed 2020-10-31 by hamster65