wled/WLED #1296: Random Cycle with segments

DarthWeber, 2020-10-29:
**Describe the bug**
Random Cycle does not work with more than one segment

**To Reproduce**
I can reproduce it with 2 segments, each different effects. Random cycle only works with one segment

**Expected behavior**
would expect that random cycle also works with mor than one segment / on selected segments

**WLED version**
 - Board Wemos D1
 - Version 0.10.2
 - Format self-compiled

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

DarthWeber, 2020-11-17:
can somebody confirm that issue or is it a problem of my setup ?

Aircoookie, 2020-11-17:
Hi!
I can confirm that this is currently a limitation of WLED.
Making random cycle and the palette transition feature available on multiple segments would require quite a bit of extra memory - I am considering at least adding it optionally though :)

blazoncek, 2023-01-25:
Random cycle woks with segments as of 0.14. I also implemented smooth palette transitions when palette changes.
It is not segment independent, though, as all segments share same underlying palette.

closed 2023-01-25 by blazoncek