wled/WLED #1298: Persist every change

yanoosh, 2020-10-29:
Hi!

I uploaded the latest version of WLED to 3 devices. Each of them works on different environments/components. I've set the same effect on all of them, not connected to led strips and start monitoring how they work with new software. As I expect they start restarting. First restart after 40min, others after 2-3h. First solution I found was to set up predefined effects. It helps but I want to choose one of 50 effects, find the ideal colour, set other parameters and forget. After restart, all my settings were gone.

I work on a similar project and have the same issue. I add a saving setting after each change. It helps because when I set all parameters they load after restart. This option can be activated by flag during compilation or in the web interface.

Similar solution works on sonoff. Each change is saved in memory, each "on" and "off" is saved in memory. My devices work about 1 year without issue (I connect all light to the sonoff devices) and I think they will work for another 5 years or more.

In my issue devices are not the main problem because there are many reasons which can restart a device: problems with electricity, hanging connection, wifi issue and many others. Some glitch during effect is acceptable but change to other effect colour is not good behaviour.

huggy-d1, 2020-10-30:
That is a lot of flash memory writing.  That's why they decided not to persist every change.
There are at least 2 ways to _persist_ changes.
1. Save to Preset 16 which gets used when wled restarts to set starting values.
2. When you are happy with current settings, perform a JSON API request to have WLED send all the settings to you.  You can use various tools to receive the JSON data to an always-on device on your network (maybe a raspberry pi?).  If that same device is monitoring to see if WLED goes offline and then comes back online, it can automatically restore the saved settings by sending them back to WLED.
Or, you can manually backup the settings and create a URL and/or JSON blob and send them any time you like - in case you wan to experiment and get back to where you were last happy enough to save the settings.

yanoosh, 2020-10-30:
I understand I can bypass my problem by predefined settings but when I lose power the lights will switch back to default presets. When I turn on the light I don't want to turn them off after restart.

Second approach requires a server which should react on restart. This solution will not work right away. After restart it receives information about the problem and then sends it back to the device. It takes some time. When u dont have a server or it will be down then auto restore will not work.

We can look for solutions but it is a simple issue which can be resolved by saving settings after each change. EEPROM should survive at least 100000 write cycles, when software saves 10 times per day it will work without problem for about 27 years, even when it saves 100 times it survives less than 3 years.

Some cases persisting will not be a good solution for example when a device will be connected to a motion sensor but then we don't need to save last settings because light state is managed by other devices.

In my case I want to connect this device to a season led strip and operate it for 3 month. Before each season I make some changes so I dont think a particular device will be used longer than 3-4 year.

huggy-d1, 2020-10-30:
I hear you and I have no control over that decision, made long ago by @Aircoookie.  In discourse posts and/or issue comments, he has stated the logical thinking behind the decision, and then started down a path that would seem to allow for it.

In other words, sometime next year, there may yet be a built-in WLED feature that could do something like that, if not exactly like that.  

Since you've already made up your mind how the solution should look, WLED is open source.  Download it and change it to be whatever you want.  If you think your changes would be of value to others, submit a Pull Request so the changes can be reviewed and possibly merged into the master branch (aka official) WLED source code.

yanoosh, 2020-10-31:
After my last comment I started thinking and researching code. I have some ideas how to do this feature. It will be done sooner than later :)

Thanks for your time.

closed 2020-10-31 by yanoosh