wled/WLED #1300: Auto reboot function?

cemasss, 2020-10-30:
Hello, it is possible to create a function in settings that will reboot ESP8266 every 2 days in midday? I noticed that ESP8266 after working for a longer period of time freezes user interface and device unreachable. So i think it would be a useful feature for the ESP8266 to reboot itself every day or every few days.

huggy-d1, 2020-10-30:
There are watchdog devices which can kick a reset signal when a signal stops toggling.
Here's one example: https://www.robotshop.com/en/watchdog-timer-module.html?gclid=EAIaIQobChMIvvCLlYPd7AIVx5-zCh0Kjgr-EAYYAiABEgKWZfD_BwE

It would probably be best if a usermod was added that toggled any pin not already in use - OR - if the datapin was guaranteed to toggle at least once every 5 minutes, for example, even if the strip was displaying all black or was off, WLED would still need to output black to ~1 pixel / minute.

Aircoookie, 2020-11-01:
@huggy-d1 's suggestion is good, you could use some external device to toggle the reset pin or the power!

A software reset should also be possible without additional code though. Just set `RB` as one of the macros (this is the reboot API command) and set it as a timed macro to go off at a specific time of day. You could also choose only some weekdays for it to reboot. Just keep in mind to never set it for what would be UTC 0:00 in your time zone (to be on the safe side do something like reboot at 1:57), or you might create a bootloop since UTC 0:00 is the time WLED has before obtaining the network time and it would immediately trigger the reboot macro.

pfeerick, 2020-11-06:
Interesting... so are you seeing this with the latest release binary @cemasss? (i.e. 0.10.2) As I noticed that my ESP8266 WLED had stopped responding just now, and was in the process of rolling back the firmware as I'm running the master branch with one or two minor changes (i.e. because I did a PR around enabling the ESP8266 to power down more, and get the benefits of PR #1245). I've just reverted to 0.10.2 to see if it is more stable. It would be interesting to catch the ESP8266 in the act, because my other ESP8266 devices have several weeks on their uptime... i.e. Tasmota devices, plus other DIY stuff ... so it's something happening in WLED, which is understandable as it's a pretty complex bit of code! ;)

blazoncek, 2022-07-25:
Watchdog timer has been implemented by #2657 , if you want to schedule regular WLED reboot it can be done using time scheduled presets.
Closing.

closed 2022-07-25 by blazoncek