wled/WLED #1310: Different segments with saved presets

dallenroot, 2020-11-01:
Could you make it so when you save a preset it saves the current segments that you have with it? I would like to make different segments do different things with the different preset button. Example: preset 1 I want the led's to flow toward the eves on my house and preset 2 to have less flow away from my eves. 

Is this possible?

Thank you!!

handsomejackuk, 2020-11-02:
i been thinking similar my self.... so for my application i want say 20 leds for the rear of my motorbike.. when i turn on motorbike wled boots and just puts on rear light...  middle  8 leds... if i press a button it will load anther preset and leave rear light on but  flash left indicators (6 leds) etc... is this possible with current release ? if i double press a button it will load different preset and flash right indicators... just a thought.... also multiple input buttons to trigger different presets at the moment its limited to only one push buton

Legsmaniac, 2020-11-03:
Just curious but why would you want to use WLED for the motorbike lights? There are custom ready made routines already out there available for indicators on Arduino using WS2812 or similar. The use of any other lights on a motorbike would be illegal, except of course privately. 🙂

Aircoookie, 2020-11-04:
This possibility is coming very soon, in 0.11 all presets will be able to store different segment configs!

closed 2021-06-04 by stale[bot]