wled/WLED #1314: Previous slave in sync relationship becomes uncontrollable

qeggs, 2020-11-04:
If a sync relationship is terminated, the WLED controller which was acting as the slave keeps the previous settings from the sync relationship and it becomes uncontrollable. The web interface for the previous slave controller appears to be online and accepting commands, but there is no response from changing any setting or attempting to turn it off. Rebooting the controller has no effect. Powering off and on will reset the controller to the default orange, but does not restore control. Cannot change brightness, patterns, or turn off.

Additionally, it appears the slave will only sync when the pattern is changed on the master, but not with any other settings like color or speed. To successfully change the brightness, the brightness must be changed with no effect, and then the pattern changed to "push" both changes to the slave controller. Sometimes the slave won't even turn off if the master is turned off. The only way to successfully turn off the slave is to first change the pattern, and then hit the off button.

**To Reproduce**
Sync two WLED controllers and get them displaying a pattern. 
Stop the sync while both are playing the pattern.
Attempt to change the pattern or turn off the previous slave controller.

**Expected behavior**
The previous slave, SHOULD turn off or the pattern SHOULD change when commanded.

**WLED version**
 - Board: NodeMCU
 - Version WLED_0.10.2_ESP8266.bin
 - Format Binary

Please ask questions or propose test cases to run. Honestly, the steps to reproduce aren't 100% clear to me.

Aircoookie, 2020-11-04:
Hi, thanks for reporting and sorry for the issue the sync causes for you! (although I suspect the uncontrollability has a different cause, maybe in wifi connectivity)
What are your settings in the `WLED broadcast` section of Sync settings, for both units?
How do you terminate the sync, from the master UI button, slave UI button, or other means?

qeggs, 2020-11-04:
Broadcast settings are default on both units:
UDP Port: 21324
Receive: brightness, color, effects all checked
Send Philips hue checked
Receive UPD realtime is checked

Also On/Off Button enabled is checked, which seems to indicate I should be able to turn off both strips from their own web page regardless of sync. My primary concern is that I could not turn my holiday lights off one night with cutting the power. The next day I figured out if I toggled the pattern and then disabled sync, it would let me turn them off and control them normally.

I terminate the sync by clicking the toggle off on the master controller. I may have tried at one point clicking the sync button ON with the slave controller, but this didn't do anything. I don't recall when I tried that, but was probably during the "zombie lights" event where I couldn't get the lights which were slaved to turn off late at night.

Note: I have had some occasional issues with spotty wifi connection to these controllers, so I find it useful that the web page indicates a connected status or gives an error if the controller can't be reached.

Aircoookie, 2020-11-04:
If `Send notifications on direct change` is checked it is indeed very strange that changing the effect would consistently cause the sync to work, while it doesn't with brightness/color.
My suspicion is that this is rather a problem with the wifi connection than with the sync itself...
`Notify twice` might also help.

A little tip: I have an outside "slave" light with terrible connection as well. Enabled NTP and set a `T=0` macro timed to turn off at 10pm. That way it turns off every night even if the sync from master is not working at the time that is turned off.

qeggs, 2020-11-05:
I just now checked `Send notifications on direct change` and `Notify twice`. I'll try this out and see if it improves the reliability of the control. I also implemented the recommended macro to turn off the lights on both the master and the slave as a backup method.

Thanks for the suggestions. This issue may be closed and I'll reopen if I have further evidence of a bug not related to spotty wifi connection.

closed 2021-06-04 by stale[bot]