wled/WLED #1315: [Important, please read] Replies from Aircoookie

Aircoookie, 2020-11-04:
Hey there my dear WLED community,

lately I must admit that I became a bit overwhelmed with the project.
That is not because I wouldn't like coding it any more, that is as awesome as always, but as you probably know, the userbase of WLED has grown about 5 times since a year ago, which I am truly happy about.

Naturally, more users means more people who are looking for help and submitting issues, posts and e-mails.
I don't like writing this, but I feel like I have to in order to be fair to you. 
In the last few months, the quality of support I was able to offer each individual has declined dramatically and I reached the point where I feel bad for many, because I wasn't able to reply for days, weeks, or even months.
The Github issue section is always at 250, even though there is an admittingly annoying bot that closes older issues automatically.
I always have 150 e-mails in my inbox, be it direct, a Github notification, or Discord messages. As I reply to one, another one comes in.

I would have wished to stay away from this point, always wanted to be the dev that responds to every single question their community might have, but I need to say that I possibly can't help you anymore.
The level of support from me you are used to is easily possible when your project has 100 users, but it becomes humanly impossible once your community grows to 4.000.
When you write a message, please do not take it personal if you don't get a response, if yours was the only message, I'd spend two hours just to help you.
I promise to still read everything, so your valuable feedback is not going to waste.

The time I can spend on WLED is very limited in my opinion. I still am a full time college student with other responsibilities.
The progress I was able to make in WLED and in parts of my personal life were far below what I would have hoped for the past few months, which I mostly attribute to spending more time interacting with all of you than actually coding.
Often I spend late nights getting the message backlog down a little, just to be too sleep deprived to do anything the next day.
As much at it hurts to potentially let some of you down, I need to change that.

Don't get me wrong, I am eternally grateful for your bug reports, feature requests and of course code contributions,
especially so to the members of my community who tirelessly offer their own time helping others!

It's just that my day only has 24 hours, like yours.
I NEED to spend my limited time bringing WLED further and making it better, which everyone will benefit from.
Chances are there's an update just around the corner that will alleviate the issue you are having as well.
Please let me get to it.

Thank you so much for reading and for your understanding!

Christian

Aircoookie, 2020-11-04:
## What does this mean for you?

### You have an issue or question about your specific setup, installation, or usermod
_For example, your LEDs flicker, you need a binary for a specific pin, or a feature doesn't work for you._

Of course I will read your message. If I believe I can help you in a few sentences, I will.
If I would need to write code or don't immediately know the cause of the issue, I most likely do not have the time, and I'm sorry for that.
Maybe a fellow community member will be able help you, but of course you should not take their valuable help for granted.
I do not take commissions, whether paid or unpaid.

### You have found a general issue or have a feature request that will benefit a lot of users
_For example, there is a reproducible issue or you have an idea for a new feature, like supporting a new protocol, or a cool new effect._

After checking that it does not already exist, feel free to open an issue! If an issue already overlaps with part of what you have in mind, comment on that instead.
Don't be concerned if no one responds, I do see your feedback and account for it in planning further development.
I might get to the issue immediately or never, it depends on how high of a priority I see it for the project. If something is wanted by a lot of users, I am more likely to regard it as useful even though I might personally think otherwise.
Therefore, adding reactions to issues or feature requests you are also interested in helps a lot!

### You want to help make WLED better by contributing code
_For example, you managed to fix an issue in WLED, added a cool feature or optimized existing code, or you are planning to do so._

Pull requests are by far the most difficult aspect of your contributions for me to say no to, as I know how much heart, soul and of course time it takes to make yourself familiar with a project and improve upon it.
Sometimes though, cleaning up, testing, and merging a PR takes more time than making the change itself, so it really depends on the type of contribution:
- The wiki is open and I always appreciate it if you choose to improve the available documentation!
- Usermod PRs are always welcome. As you are responsible for them and since they do not modify the WLED core, I can merge them without thorough review and testing.
- If you make a change primarily for yourself, but believe others would benefit too, feel free to open a PR!
Just please don't be disappointed if the PR doesn't get merged or commented on for a long time, either it is not a good fit at the moment or it requires changes too substantial to merge immediately.
- If you want to make a change to make WLED better even though you personally wouldn't benefit from it - don't. If you are about to pour a significant investment into creating a PR, ask first in an issue whether it would be helping out.
The thing I hate most is hard work going to waste, so I'd like to spare you from that.
- PRs are more difficult to review the more files are changed - if you want to make multiple changes, submit multiple PRs with only one feature added at a time. This ensures at least some of your work makes it in faster.
- Lastly, if I need help, I will ask for help :) Soon I will start labeling issues with the "help wanted" label, they are a great fit for PR fixes. I'd recommend a small note in the issue to let others know you are working on the feature - otherwise it could happen that two people work on the same thing.

Thank you so much for being a valuable part of our community!

nibblerrick, 2020-11-04:
I think this is an important step when the project and especially the userbase grows.
I hope that noone will be angry at you for this as you made this already in this state great project.
Personally I like that you want to keep working on it and not just abandone it because of the amount of supportrequests you get.
Thanks for letting us know and I hope you get on trackk with organizing these things, that's always a difficult task.

CMaxim77, 2020-11-04:
Thanks for all you have done with this project. And I completely understand that a wildly popular project like this can become out of hand from the general public. I'm glad that you are still committed to evolving this software and I'm excited to see what else comes in the future. Your first priority should be your schooling and the rest can be done if/when you have time to do so. I don't know if people always understand the amount of work these projects take and I think sometimes they forget that this is not a profitable thing for you. I'm not a coder but I'll happily, quietly wait for updates as you can get them in. I would encourage everyone to consider donating anything they can to help compensate you for the amount of efforts you put in here. Thanks again for everything you do!

sansillusion, 2020-11-04:
Thanks fornall your great work. I think that all the old user will remember
you extreme dedication to your project and all the support you give. All
new user should be understanding of this too (most project don't have that
level of support anyway). And anyone that complains are not worth the
effort. Thanks for your great work and I am sure you have a bright future
ahead of you ! Such a great coder and it shows that you are a great human.
I wish the best for you.

jdavis7765, 2020-11-05:
Just want to thank you for all your hard work and devotion you have put into this Aircookie. It is truly an awesome project and has opened me to the world of digital LEDs (I was already booked on LEDs anyway). I now have too many LEDs strips in my house lol. There are plenty of resources out there to help people and it seems there is always someone available to help with them or me at times ☺️. 

Again, thanks for all your hard work and time on this. It is truly an awesome project and you should be proud of yourself for it!

THATDONFC, 2020-11-06:
We appreciate you! Thank you for bringing this amazing project to reality and for continuing to build on it, even with a regular life. I know how hard it can be to manage so many tasks and also be a people pleaser. It's impossible to do it all. You are doing a killer job, and I am grateful to have WLED and the WLED community that you started.

As always, I am here to help. If you need anything, from debugging, to documentation, to answering questions from new members, I'm here. I do my best to answer everyone in the discord that has gone unanswered. If you think there are other places I could focus my attention to help, please let me know. I'm also going to turn on notifications for the PR's and issues here on gh so I can try to help others that might not be on discord. 

Thanks for everything you do!

fensoft, 2020-11-13:
maybe open a tipeee or utip ? i'm pretty sure a lot of us are ready to donate ...

Operation760, 2020-11-14:
If there is anything we can do offload some of the burden, especially outside of PR's etc, specifically administration-wise, don't hesitate to say so. Speaking for all of us, I feel the service you've done is awesome, especially for a part time college kid. Color me impressed.  

Having spearheaded over 200 IoT full-stack, high-volume mobile cellular/wifi/BT/GNSS telemetry products in the last decade, please do send me your CV when/if you have ambitions beyond college in embedded, middle-ware or higher up the stack. 

In the meantime, if I can do some project management, scheduling, roadmap development, etc, I'm in kid. 👍

kernowquack, 2020-11-18:
As a newbie WLED user I think the project is great and just spending a little time looking around the wiki and current issues. it shows that you care about the project and its users. Keep coding and I'm sure the community will step up and help each other out

otherguy, 2020-11-30:
@Aircoookie I'm using this thread to say thank you for building such an amazing piece of software. Truly among the best Open Source projects, from code to documentation to presentation. Amazing! Thank you again and stay safe out there!

asknoone, 2020-12-03:
Hey @Aircoookie. Adding to the Home Automation community like this amazing.

As far as I can tell you don't have Patreon account? You should set one up, I'm sure there will be people who will be prepared to support you on a monthly basis rather than just a PayPal link. Sure, money doesn't give you more than 24 hours in a day 🙂 but it helps...and you deserve it.

Aircoookie, 2020-12-03:
@asknoone thank you very much for the kind words :)
I will consider making a Patreon or Github Sponsors at some point! PayPal was the easiest way for now.

otherguy, 2020-12-03:
@Aircoookie do set up GitHub sponsors if you can! It’s nicely integrated with GitHub and at least you know everyone using your software has a GitHub account ;)

teocreo, 2021-01-06:
> 
> 
> If there is anything we can do offload some of the burden, especially outside of PR's etc, specifically administration-wise, don't hesitate to say so. Speaking for all of us, I feel the service you've done is awesome, especially for a part time college kid. Color me impressed.
> 
> Having spearheaded over 200 IoT full-stack, high-volume mobile cellular/wifi/BT/GNSS telemetry products in the last decade, please do send me your CV when/if you have ambitions beyond college in embedded, middle-ware or higher up the stack.
> 
> In the meantime, if I can do some project management, scheduling, roadmap development, etc, I'm in kid. +1

Operation760 ... how is possible contact you ??

mghaff007, 2021-03-01:
I ran across WLED while looking for a controller for my boy to try and get him into our sport. I myself have solely been using WLED for everything! I feel bad that i am no longer searching the internet and playing with others projects, but i haven't had the need to anymore... All being said @Aircoookie THANK YOU for giving me more time to work on my LED projects than trying to figure out how to make them work. Because of your software THEY ALWAYS WORK!!! SO MANY THANKS!

Eddiiie, 2021-12-10:
College is over rated.

jmerc77, 2023-07-21:
hay aircookie! incase this motivates you a little, #2496 (regarding a wifi mdns issue) may have accidentally fixed! the issue has not been closed yet, but i found that a router reset fixed a similar issue i was having so not the fault of WLED! i feel it can be closed.

BitVenturesUSA, 2023-07-29:
Congratulations on the overwhelming success of WLED! You are in a great spot to start scaling your app and unleashing the unrealized power it has.

I am sure you get offers for help with the coding and design of the app frequently but where I could potentially help you is with building a structure around the project that allows it to be clearly defined and standardized with a road map and your vision as to where you would like the project to be in 1-3 years. That would allow a business to formed around it and open up a world of resources at your disposal. You have dropped something into the space that should be licensed and have strategic partnerships with OEMs ISVs and any other entities that would be mutually beneficial. Couple all that with wide distribution and marketing with key individuals working as a collaborative team responsible for it's continued success. 

I know that seems like a massive amount of work and startup capital requirements in the hundreds of thousands but it's not. Modern platforms and AI/ML Technology have flipped the script and upended the process and cost of bringing an idea to market. I'm confident it could fund portions of it's own growth while any capital requirements are managed through investors and other financial instruments. For the record, I'm a lifelong advocate of open source software and am in no way suggesting the project would have to abandon that ethos. 

I see a lot more than a led light controller here. I see something disruptive that could result in some significant financial gains and recognition that would also benefit your future endeavors.

if you want to chat about it, feel free to reach out!
sean@four32.io