wled/WLED #1318: Green and Blue mixed up with ws2801

tilmanschwaner, 2020-11-05:
Green and Blue is mixed up in WS2801-Image

cschuhknecht, 2020-11-06:
I don't know what the "-Image" means, but if the strip shows wrong colors because green and blue are swapped, you can change the color order in the LED settings just above "Defaults".

huggy-d1, 2020-11-06:
Since there is not exact specification what the color order should be, so LED strip manufacturers choose what suits their capabilities or perhaps their notion of what the correct order should be.

I prefer RGB, but can see reasoning for BGR (English, alphabetical), or any other order for a different language's preferred ordering.

WLED lets you match pixel color order to the LED strip connected to it.  I suspect the bug is not with WLED, but with the Issue poster not realizing this capability existed.

jezmck, 2020-11-09:
@tilmanschwaner, can you please confirm that you're able to use the **Color order** setting in **Config > LED Preferences**?

[image]

If so, please mark this issue as closed.

closed 2020-11-20 by Aircoookie

tilmanschwaner, 2020-12-22:
I haven´t seen this option, it works now, thank you!